ARMmbed/mbed-os #8378: STM32F4: Disabling one of several InterruptIn objects connected to different pins disables other interrupts

Wlad-Fox, 2018-10-11:
### Description

Target MCU: STM32F446RE
If you initialize two InterruptIns on pins PB14 and PB15, after deleting one, second one will stop working too. It happens because:

1. In st32f446 devices external interrupts on pins from 10 to 15 are handled by one interrupt vector EXTI15_10. 
2. When destroying InterruptIn the function void gpio_irq_free(gpio_irq_t *obj) disables interrupt vector connected to pin. So if it disables EXTI15_10, any of InterruptIn objects connected to PB10-PB15 will stop working.

### Issue request type

[ ] Question  
[ ] Enhancement  
[X] Bug

LMESTM, 2018-10-11:
@Wlad-Fox could you check whether the issue is addressed by #8037 ?

Wlad-Fox, 2018-10-15:
@LMESTM Yes, it is, thank you, I'll close the issue.

closed 2018-10-15 by Wlad-Fox