ARMmbed/mbed-os #8396: Fix python module versions within requirements.txt

cmonr, 2018-10-11:
### Description

After finding and solving an issue with colorama being updated (resolved with https://github.com/ARMmbed/mbed-os/pull/8390), @studavekar brought up a good comment that we should do the same with other packages.

Requirements.txt should be fixed to specific versions, and have a process to update versions on a regular bases. This issue is for the first part of the problem, with @ARMmbed/mbed-os-test already working on the second.

### Issue request type

[ ] Question  
[x] Enhancement  
[ ] Bug

OPpuolitaival, 2018-10-12:
I think that this make things a lot easier because now it is not easy to track when tools versions are updated. When there are specific version then PR itself is used to test the new version of the tool

cmonr, 2018-10-16:
Aaand it looks like we just got hit again by not having versions fixed/pinned.

Last working master build: https://travis-ci.org/ARMmbed/mbed-os/builds/442254105
The next build after: https://travis-ci.org/ARMmbed/mbed-os/builds/442300780

(urllib3 was updated from 1.23 to 1.24)

yennster, 2018-10-23:
@cmonr Does https://github.com/ARMmbed/mbed-os/pull/8446 fix this issue?

studavekar, 2018-10-23:
@yennster this issue is to review https://github.com/ARMmbed/mbed-os/blob/master/requirements.txt and update to latest working version. As of now, most of packages say ```>=``` which are causing failures as latest version of module may not be compatible with other modules used.

problem seen
1) https://github.com/ARMmbed/htrun/issues/222
2) https://github.com/ARMmbed/mbed-os/pull/8446

cmonr, 2018-10-23:
@yennster Not completely.

#8466 was the immediate fix needed to make master green again, but this ticket is a placeholder to make sure that we update requirements.txt so that _all_ module versions are updated accordingly.

Feel free to assign the internal ticket to me.

yennster, 2018-10-24:
@cmonr Moved to IOTCORE and assigned it to you (IOTCORE-571)

closed 2018-11-01 by cmonr