ARMmbed/mbed-os #8399: STM32F1: HAL_RTC_SetAlarm_IT corrupts system time

Arjun765, 2018-10-12:
### Description
Trying to set RTC wakeup by calling `HAL_RTC_SetAlarm_IT` corrupts system time. 
Related: #8213 #8191 

### Issue request type

[ ] Question  
[ ] Enhancement  
[ x] Bug

elm3, 2019-01-07:
@ARMmbed/team-st-mcd

LMESTM, 2019-01-07:
@Arjun765 
By calling  HAL_RTC_SetAlarm_IT you're bypassing mbed Time layer so this is not recommended to do so. You should rather use mbed APIs. 

mbed is designed so that you can program any timer and mbed-os is in charge of programing RTC or low power tickers to wake-up the system when needed, you don't have to program an RTC alarm explicitly. 

There have been many similar discussions for instance here : #7156

TacoGrandeTX, 2019-02-11:
@Arjun765 We're going to close this as a non-issue based on the comments from @LMESTM.  If you have a problem using the Mbed API please raise a new issue.

closed 2019-02-11 by TacoGrandeTX