ARMmbed/mbed-os #8428: artifact_name option in CMake exporter not changing name

vidavidorra, 2018-10-15:
### Description

Running the `mbed export -i cmake_gcc_arm` when the `artifact_name` is configured in the `mbed_app.json` the project name is still the one specified by the first `--source` argument rather than the value set by the `artifact_name` option.
See comments [6585 theotherjimmy comment](https://github.com/ARMmbed/mbed-os/issues/6585#issuecomment-382881700) and [6585 vidavidorra comment](https://github.com/ARMmbed/mbed-os/issues/6585#issuecomment-429222889).

This issue can be reproduced by running the export on any `mbed-os-example` with the `-i cmake_gcc_arm` option and with the `artifact_name` option set in the `mbed_app.json` and at least one `--source` given. Output can be seen in the `CMakeLists.txt` where `TARGET` is the name of the first --source argument rather than the `artifact_name`.

Using mbed-os-5.10.0, mbed-cli 1.7.5.

The `artifact_name` does not work as expected. I've only seen this with the CMake exporter since that's the only one I use, so I don't know whether this also affects other exporters or build.

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

vidavidorra, 2019-01-17:
Any update on this? @theotherjimmy perhaps?

This is still not working in 5.11.1. `artifact_name` in `mbed_app.json` doesn't seem to change anything in the output name and the `--artifact-name` argument doesn't seem to be supported in `mbed-export`

theotherjimmy, 2019-01-17:
Hi @vidavidorra,

I have not worked on this. I take it that you use the cmake exporter pretty frequently. @kegilbert could you look into this for me? I think the steps to reproduce are pretty solid, and you could look in the cmake exporter for how the `TARGET` is populated and see if `config.name` would work instead.

kegilbert, 2019-01-31:
Sure thing, sorry for missing this earlier!

vidavidorra, 2019-02-01:
@kegilbert No worries. @theotherjimmy Yes, I use it all the time!

It would be great if both the `artifact_name` in `mbed_app.json` and `--artifact-name` option to `mbed-export` can be added. In that case, the option would overwrite a value set by `mbed_app.json`, but I'm not sure how this works with other tools.

kegilbert, 2019-03-01:
@vidavidorra https://github.com/ARMmbed/mbed-os/pull/9817 should help out, let me know if you have any further issues!

closed 2019-05-22 by ciarmcom