ARMmbed/mbed-os #8431: Current map file is named old

JammuKekkonen, 2018-10-15:
### Description

Map file is renamed during building as 'old', even though it's the result of the most recent build.

### Issue request type
    [ ] Question
    [ ] Enhancement
    [X] Bug

JammuKekkonen, 2018-10-15:
@theotherjimmy 

FYI: @TeroJaasko @korjaa

kjbracey-arm, 2018-10-15:
Yes, would make much more sense to attempt to rename ".map" as ".map.old" as the first step of a build, not the last.

vidavidorra, 2018-10-18:
I think that this is due to running memap. See #7590 I haven't created a separate ticket but now you have :)

Thought I reference the PR here, scroll down in the comments to see my comment and a proposed fix. If it's the same it should actually append `.old`

theotherjimmy, 2019-01-17:
@JammuKekkonen I fixed this a while back. Let me find the PR.

theotherjimmy, 2019-01-17:
https://github.com/ARMmbed/mbed-os/pull/8779 should have fixed this. @JammuKekkonen @vidavidorra Does this behavior change in 5.11.1?

vidavidorra, 2019-01-17:
Yes, the file now stays in the file system, so would indeed fix this ticket.

@theotherjimmy As for the PR you referenced, I read through it and you note that the migration is now moved to the toolchain. I'm using the exporter and have added a custom command in my CMake project file to run the memap after a build, since it is not ran automatically for CMake export at least. With moved to the toolchain, do you mean that it is handled by `mbed compile` for example? If so, that would explain why my memap doesn't do differential since there would be no old file to compare with.

theotherjimmy, 2019-01-17:
> With moved to the toolchain, do you mean that it is handled by `mbed compile` for example?

Yes.

> If so, that would explain why my memap doesn't do differential since there would be no old file to compare with.

Yes, it does explain it.

JammuKekkonen, 2019-01-22:
This issue is fixed.

closed 2019-01-22 by JammuKekkonen