ARMmbed/mbed-os #8438: CMSIS-RTOS Error: C/C++ library libspace not available

bqam-ublox, 2018-10-16:
### Description

I faced the following issue: **Standard C/C++ library libspace not available**. To resolve this, I changed value of **OS_THREAD_LIBSPACE_NUM** from 4 to 6. This probably increases the number of threads that access the C/C++ library and hence resolves the issue.

However, I could not find a way to define the **OS_THREAD_LIBSPACE_NUM** through mbed_app.json or any other configuration file since [OS_THREAD_LIBSPACE_NUM](https://github.com/ARMmbed/mbed-os/blob/master/rtos/TARGET_CORTEX/rtx5/RTX/Config/RTX_Config.h#L537) is defined inside an if condition and its scope is also limited within RTX_Config.h. 

Is there any way to make it work without making changes in mbed source/rtos file? 

### Issue request type
    [x] Question
    [ ] Enhancement
    [ ] Bug

bqam-ublox, 2018-10-16:
CC: @RobMeades , @fahim-ublox

fahim-ublox, 2018-10-16:
@screamerbg

0xc0170, 2018-11-22:
Quick one would be to define a macro via config `OS_THREAD_OBJ_MEM` , this should overwrite the default one.

You can create a new config value and set macro name to `OS_THREAD_OBJ_MEM` and assign a value you need.

closed 2019-01-18 by TacoGrandeTX