ARMmbed/mbed-os #8466: Incorrect page size returned from Flash IAP

LaurentLouf, 2018-10-18:
### Description

- Target STM32F103RCT6
- GNU ARM 7-2017-q4-major
- mbed-cli 1.7.3
- 603c4f930 (HEAD -> master) Merge pull request #7851 from geky/fix-littlefs-mount-cleanup

A minimal code to reproduce the issue would be : 
```
#include "mbed.h"

// Internal flash interface
FlashIAP internal_flash;

int main() {
    // Initialize internal flash interface
    internal_flash.init();
    printf("Flash start %d, size %d, page size %d\r\n", internal_flash.get_flash_start(),
           internal_flash.get_flash_size(), internal_flash.get_page_size());
}
```
This issue being that I've got the following response on my target : 

> Flash start 134217728, size 262144, ***page size 4***

So the flash start and size are coherent with my target, but the page size isn't at all. According to the doc of Flash IAP (https://os.mbed.com/docs/v5.8/mbed-os-api-doxy/classmbed_1_1_flash_i_a_p.html), the value returned is expressed in byte and I highly doubt that I have page sizes of 4 bytes (should be 2kbytes if I recall correctly). I've had to adapt a few files to support my target, but it's been a while and I don't have any problems with the changes made, so I assume that they are mostly correct. So could it be an error with the API of Flash IAP ? 

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

LMESTM, 2018-10-18:
@LaurentLouf 
the page size definition in mbed is the program page size, which is:
```
     *  The page size defines the writable page size
     *  @return Size of a program page in bytes
```
and your STM32 target allows to write the flash down to 4 bytes at a time (you cannot write 1, 2 or 3 bytes only, it has to be be at least 4 bytes), which corresponds to the definition of the page size in mbed.

LaurentLouf, 2018-10-18:
@LMESTM Well thanks for the quick reply ! I thought about that shortly after posting the question and I have to agree that the terminology is a bit confusing. Thanks for the information about the different sector sizes, I'll have to check that carefully.

closed 2018-10-18 by LaurentLouf