ARMmbed/mbed-os #8467: [STM32] Mutex lock fail, if IRQ are disabled

pmancele, 2018-10-18:
### Description

Hello,
We are working on  a `STM32L496RG` and also reproduced the issue on a `NUCLEO_L476RG`. This may apply to all STM32L4 familiy (or more ?)

Since the #6872 merge request, each call to `Mutex::lock()` function triggers an mbed error if a call to `__disable_irq()` has been done previously, which print the following message:

```
++ MbedOS Error Info ++
Error Status: 0x80020115 Code: 277 Module: 2
Error Message: Mutex lock failed
Location: 0x8002F09
File:Mutex.cpp+62
Error Value: 0xFFFFFFFA
Current Thread: Id: 0x200024AC Entry: 0x800329F StackSize: 0x1000 StackMem: 0x200024F0 SP: 0x20003380 
For more info, visit: https://armmbed.github.io/mbedos-error/?error=0x80020115
-- MbedOS Error Info --
```

For example, the following code triggers the error, as `SPI::write()` calls the `lock()` function:

```cpp
__disable_irq();

spi.write(0x00);

__enable_irq();
```

Is it a bug ?

This was done like this in a driver written by Semtech for the SX1280 LoRa component. I removed the `__disable_irq()` and it is now working, but I assume `disable_irq()` was called for a reason..

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

kjbracey-arm, 2018-10-18:
This would have been failing on debug builds for some time, but the check is now in default develop builds too.

The SPI API is designated as "thread safe" -  "safe for use from multiple threads; operation is protected by an RTOS primitive and can be used from multiple threads, but will cause problems if used from an interrupt service routine." ([old link](https://docs.mbed.com/docs/mbed-os-handbook/en/5.2/concepts/thread_safety/) - can't find in current docs)

If you really need to do this from IRQ, you could override the locking to make it interrupt safe - make a class that inherits from `SPI` and overrides the `lock` and `unlock` functions to do irq disable and enable. Not sure that's 100% safe, but probably works on most platforms. Even better would be to move the SPI work to a thread, or maybe one of the global event queues.

If you're not using it from interrupt, I don't see any reason for the disable IRQ on its own, unless it's also protecting some of your own data structures from an IRQ routine.

pmancele, 2018-10-18:
@kjbracey-arm Thank you for you detailled andwer.

Yes, I'm used to use EventQueue to process my work out of ISR context. But the code that I reported is not from me but from the [SX1280 driver](https://os.mbed.com/teams/Semtech/code/SX1280Lib/). Into the `WakeUp` function:

```cpp
void SX1280Hal::Wakeup( void )
{
    __disable_irq( );
  
    RadioNss = 0;
    RadioSpi->write( RADIO_GET_STATUS );
    RadioSpi->write( 0 );
    RadioNss = 1;
   
    // Wait for chip to be ready.
    WaitOnBusy( );
 
    __enable_irq( );
}
```

I think we can close the issue.

kjbracey-arm, 2018-10-18:
A possible guess is that the intent there may be to stop the driver's interrupt handler running due to the radio generating an interrupt during the "wakeup"

If that is necessary, might want to replace it with a disable of the relevant InterruptIn.

pmancele, 2018-10-18:
Indeed, I think it's the objective, and thank you for the suggested fix.

closed 2018-10-18 by pmancele