ARMmbed/mbed-os #8470: Backtrace functionality in mbed_die

daniel-cesarini-tridonic-com, 2018-10-18:
### Description

I would like to know whether backtrace functionality is expected to be available in mbed-os anytime soon, as it looks like the related task in uvisor does not proceed anymore: https://github.com/ARMmbed/uvisor/issues/70#issuecomment-429778663 .

Related issues: 
* https://github.com/ARMmbed/mbed-drivers/issues/154
* https://github.com/ARMmbed/mbed-os/issues/3131

### Issue request type

    [X] Question
    [ ] Enhancement
    [ ] Bug

FYI: @MarceloSalazar @adbridge 

Internal: @javier-moreno-tridonic-com @markus-becker-tridonic-com @andreas-myka-tridonic-com

daniel-cesarini-tridonic-com, 2018-10-18:
As an additional information, we started to look into the usage of ```MBED_CALLER_ADDR()``` at least for having some basic information about who is calling the mbed_die():

```printf("%s called from %p\n", __func__, MBED_CALLER_ADDR());```

Is there any better approach? 
Is there any other plan to have some similar functionality in mbed_die and other functions available? It would be quite handy when hunting for bugs, especially if they are difficult to reproduce systematically.

kjbracey-arm, 2018-10-18:
`mbed_die` should normally be invoked via `mbed_error` or `error` or `MBED_ASSERT`, which do give caller information.  If there are any raw calls to `mbed_die`, they should probably be changed to use one of those higher level ones.

kjbracey-arm, 2018-10-19:
Still agree I'd like the potential for more backtrace. 128 bytes of stack would not go amiss.

daniel-cesarini-tridonic-com, 2018-10-19:
@kjbracey-arm can we follow on this topic in this issue, or you prefer a new issue?

linlingao, 2019-04-30:
Reopen and change it to enhancement.

closed 2019-10-31 by ciarmcom