ARMmbed/mbed-os #8472: Compilation issue with SD component test for small RAM targets

jeromecoutant, 2018-10-18:
### Description

Some SD component tests need more RAM than other greentea tests :

- components-storage-blockdevice-component_sd-tests-filesystem-files : needs between 35K and 50K RAM depending on target and tool chain

- components-storage-blockdevice-component_sd-tests-filesystem-dirs : needs between 20K and 35K RAM depending on target and tool chain

- components-storage-blockdevice-component_sd-tests-filesystem-seek : needs between 20K and 35K RAM depending on target and tool chain

Maybe some rework can be done ?
Or some RAM check can be added in order to skip compilation ?
  #if (RAM_SIZE) < 50K
  #error [NOT_SUPPORTED] test needs bigger RAM

Example of compilation issue  :

````
Build failures:

  * NUCLEO_F091RC::GCC_ARM::COMPONENTS-STORAGE-BLOCKDEVICE-COMPONENT_SD-TESTS-FILESYSTEM-FILES
        [DEBUG] Return: 1
        [DEBUG] Errors: c:/program files (x86)/gnu tools arm embedded/6 2017-q2-update/bin/../lib/gcc/arm-none-eabi/6.3.1/../../../../arm-none-eabi/bin/ld.exe: ./BUILD/tests/NUCLEO_F091RC/GCC_ARM/./components/storage/blockdevice/COMPONENT_SD/TESTS/filesystem/files/files.elf section `.bss' will not fit in region `RAM'
        [DEBUG] Errors: c:/program files (x86)/gnu tools arm embedded/6 2017-q2-update/bin/../lib/gcc/arm-none-eabi/6.3.1/../../../../arm-none-eabi/bin/ld.exe: region RAM overflowed with stack
````

### Issue request type

    [x] Question
    [ ] Enhancement
    [ ] Bug

linlingao, 2018-10-22:
What do you gain by failing the build at compilation time rather than link time?
I also don't think adding RAM_SIZE is foolproof. Since RAM usage is just estimation at certain point in time. As code size grows, it's not feasible to keep track of how much more RAM is needed all the time. Adding a check at compilation time can be pretty misleading.

closed 2018-10-22 by linlingao

jeromecoutant, 2018-10-23:
> What do you gain by failing the build at compilation time rather than link time?

Difference is the result...

#error [NOT_SUPPORTED] :
- stops immediately the compilation
- skips the tests

Link issue :
- takes time for compilation
- declares the test as BUILD_FAILURE

jeromecoutant, 2018-10-23:
And what about the first part of the question ? Maybe some tes rework can be done in order to decrease bss region size ?

Thx

@geky @davidsaada