ARMmbed/mbed-os #8481: Unable to register for wifi status updates with Realtek 8195

michalpasztamobica, 2018-10-19:
### Description

Target: RTL8195
Toolchain: GCC_ARM 
Tools: mbed-cli 1.7.2 on Windows
mbed-os SHA: e94258204
Steps to reproduce:
1) Try to use the interfaces defined in targets/TARGET_Realtek/TARGET_AMEBA/sdk/common/api/wifi/wifi_ind.h.
For example, I modified the RTWInterface constructor to call the following:
```
init_event_callback_list();
wifi_reg_event_handler(WIFI_EVENT_STA_DISASSOC, myEventHandler, this);
```
2) run compilation:
`$ mbed compile --target REALTEK_RTL8195AM --toolchain GCC_ARM `
3) Observe linker errors:
```
[ERROR] BUILD/REALTEK_RTL8195AM/GCC_ARM/targets/TARGET_Realtek/TARGET_AMEBA/RTWInterface.o: In function `RTWInterface':
C:\***\arm\mbed-os/.\targets\TARGET_Realtek\TARGET_AMEBA/RTWInterface.cpp:138: undefined reference to `init_event_callback_list()'
C:\***\arm\mbed-os/.\targets\TARGET_Realtek\TARGET_AMEBA/RTWInterface.cpp:139: undefined reference to `wifi_reg_event_handler(unsigned int, void (*)(char*, int, int, void*), void*)'
collect2.exe: error: ld returned 1 exit status
```

BACKGROUND:

Why I am doing this is because I want to be able to receive Wifi status update (disconnected, connecting, local_up, global_up) from Realtek HW. Currently, I am only getting them during connection and I assume they are actually triggered by the connect() procedure and not by events coming from Wifi.

I downloaded the [Realtek Ameba SDK](https://www.amebaiot.com/en/?s2member_file_download=sdk-ameba-v4.0b_without_NDA_GCC_V1.0.0.zip) and after extracting it I actually found the definitions of the functions. They are not provided with the mbed-os source code.
I also examined the provided binary lib (/mbed-os/targets/TARGET_Realtek/TARGET_AMEBA/TARGET_MCU_RTL8195A/device/TOOLCHAIN_GCC_ARM/lib_wlan_mbed_gcc.a) using nm and objdump on my Linux machine:

```
$ objdump --syms lib_wlan_mbed_gcc.a | grep init_event_callback_list
00000000         *UND*    00000000 init_event_callback_list
00000000 l    d  .text.init_event_callback_list    00000000 .text.init_event_callback_list
00000000 l       .text.init_event_callback_list    00000000 $t
0000000c l       .text.init_event_callback_list    00000000 $d
00000001 g     F .text.init_event_callback_list    00000010 init_event_callback_list

$ nm lib_wlan_mbed_gcc.a | grep init_event_callback_list
         U init_event_callback_list
00000001 T init_event_callback_list
```

I am a bit worried about the fact that the function is reported as undefined.

Right now it seems to me that there is a problem with the provided lib_wlan_mbed_gcc.a library.
Please, correct me if I am not using the events API correctly.

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

SeppoTakalo, 2018-10-19:
@ARMmbed/team-realtek Please review.

samchuarm, 2018-10-26:
Hi @AdamZhang0124 , can you please confirm if the function reported is supported in Mbed OS with Realtek SDK?

M-ichae-l, 2018-11-15:
Please refer to #8750 for the fix.

samchuarm, 2018-11-20:
Since the fix has been identified, can we close this issue? Thanks.

SeppoTakalo, 2018-11-20:
Can be closed once the fix has been merged.

closed 2018-11-26 by 0xc0170