ARMmbed/mbed-os #8490: Ethernet connect fails with LWIP in mbed os

UsmanKhan0392, 2018-10-22:
### Description
**Ethernet connect fails with LWIP in mbed os**

I'm using an mbed enabled board(ublox_030) for development, i need to run an Ethernet application on it.

I tried to create a connection by using the following code:

```
`network = new EthernetInterface();
    network->connect();
    // Show the network address

    const char *ip = network->get_ip_address();
    printf("IP address is: %s\n", ip ? ip : "No IP");`
```

Normally it should work, But it fails in the LWIPInteraface class's bringup API at osSemaphoreAcquire, returning a timeout error.

```
if (!netif_is_link_up(&netif)) {
    if (blocking){
        if (osSemaphoreAcquire(linked, 15000) != osOK){
            if (ppp){
                (void) ppp_lwip_disconnect(hw);
            }
            return NSAPI_ERROR_NO_CONNECTION;
        }
    }
}

```
### Issue request type

Any reason why I might be getting a timeout from osSemaphoreAcquire? I tried increasing the timeout too but in vain. If someone could help me with it, would be appreciated. Thanks in advance.
    [x ] Question
    [ ] Enhancement
    [ ] Bug

kjbracey-arm, 2018-10-22:
If it's that semaphore, it means that it doesn't think the Ethernet cable is physically connected, or it's failing to autonegotiate with the hub. It hasn't reached the stage of doing IP yet.

I don't think you need to change any links or anything on that board to enable Ethernet, so I'd recheck your physical connections and make sure you're seeing the connection light.

kjbracey-arm, 2018-10-24:
Could you showing the entire application including mbed_app.json? This is such a fundamental failure, it seems like it has to be an application/config problem, or something U-blox C030 specific. Also, precisely which version of mbed OS?

UsmanKhan0392, 2018-10-24:
@kjbracey-arm 
I'm using the following version : mbed-os-5.10.0-rc3
And below is the application code 

```
#include "mbed.h"
#include "EthernetInterface.h"
 
int main() {
    printf("Start Application\r\n");
    EthernetInterface eth;
    eth.connect();
// Show the network address
    const char *ip = eth.get_ip_address();
    printf("IP address is: %s\n", ip ? ip : "No IP");

    const char *mac = eth.get_mac_address();
	printf("MAC address is: %s\n", mac ? mac : "No IP");

    //eth.init(); //Use DHCP
    //eth.connect();
    //printf("IP Address is %s\n", eth.getIPAddress());
    //eth.disconnect();
    // Open a socket on the network interface, and create a TCP connection to mbed.org
    TCPSocket socket;
    socket.open(&eth);
    socket.connect("www.arm.com", 80);

    // Send a simple http request
    char sbuffer[] = "GET / HTTP/1.1\r\nHost: www.arm.com\r\n\r\n";
    int scount = socket.send(sbuffer, sizeof sbuffer);
    printf("sent %d [%.*s]\n", scount, strstr(sbuffer, "\r\n")-sbuffer, sbuffer);

    // Recieve a simple http response and print out the response line
    char rbuffer[64];
    int rcount = socket.recv(rbuffer, sizeof rbuffer);
    printf("recv %d [%.*s]\n", rcount, strstr(rbuffer, "\r\n")-rbuffer, rbuffer);

    // Close the socket to return its memory and bring down the network interface
    socket.close();

    // Bring down the ethernet interface
    eth.disconnect();
    printf("Done\n");

    while(1) {}
}

```

targets.json file is shown below
```

    "UBLOX_C030": {
        "inherits": ["FAMILY_STM32"],
        "supported_form_factors": ["ARDUINO"],
        "core": "Cortex-M4F",
        "supported_toolchains": ["GCC_ARM", "ARM", "IAR"],
        "extra_labels_add": ["STM32F4", "STM32F437", "STM32F437VG", "STM32F437xx", "STM32F437xG", "STM_EMAC"],
        "config": {
            "modem_is_on_board": {
                "help": "Value: Tells the build system that the modem is on-board as oppose to a plug-in shield/module.",
                "value": 1,
                "macro_name": "MODEM_ON_BOARD"
            },
            "modem_data_connection_type": {
                "help": "Value: Defines how the modem is wired up to the MCU, e.g., data connection can be a UART or USB and so forth.",
                "value": 1,
                "macro_name": "MODEM_ON_BOARD_UART"
            }
        },
        "macros_add": ["MBEDTLS_CONFIG_HW_SUPPORT", "HSE_VALUE=12000000", "GNSSBAUD=9600"],
        "overrides": {"lse_available": 0},
        "device_has_add": ["ANALOGOUT", "EMAC", "SERIAL_FC", "TRNG", "FLASH"],
        "public": false,
        "device_name": "STM32F437VG",
        "bootloader_supported": true,
        "overrides": {
            "network-default-interface-type": "ETHERNET"
        }
    },
    "UBLOX_C030_U201": {
        "inherits": ["UBLOX_C030"],
        "release_versions": ["5"]
    },
    "UBLOX_C030_N211": {
        "inherits": ["UBLOX_C030"],
        "release_versions": ["5"]
    },
    "UBLOX_C030_R410M": {
        "inherits": ["UBLOX_C030"],
        "release_versions": ["5"]
    },
```

kjbracey-arm, 2018-10-24:
Thanks - and do you have a `mbed_app.json` in the application folder?

kjbracey-arm, 2018-10-24:
@ARMmbed/team-ublox - any suggestions? Any known C030 traps/issues?

UsmanKhan0392, 2018-10-29:
Ethernet standard mbed-os tests runs fine on the board and nothing wrong with simple Ethernet application either.

But In my application, I'm using make build system for compilation.
After including all the necessary flags, include paths and object files obtained after exporting the project for my target.

And now when I try to run the same application, it fails to get an IP address.
My application code basically never enters **netif_link_irq** and **netif_status_irq** , even though they're set to callbacks by **netif_set_link_callback**.

kjbracey-arm, 2018-10-29:
Sounds like a potential exporter problem then? I think you'll need to give more details - is this the result of a standard `mbed export`? Sorry, I'm not really familiar with project export, so someone else may need to advise.

UsmanKhan0392, 2018-10-29:
When I exported the a standalone mbed Ethernet application for my target and used **make** to build it, it ran fine and obtained an ip successfully.
 
Now, In my My project (which comprises of various other components , mbed-os being one of them) I included all the flags and include paths as was in the exported project, it failed to obtain IP.

I don't see any difference in both projects apparently (have the same include paths, flags ,object files and piece of code for Ethernet initialization and connection) and I was expecting them to behave alike, but something's causing problem which I can't identify.

kjbracey-arm, 2018-10-29:
If you have "other components", then maybe some issue attaching the IRQ handler? It appears the `STM32_EMAC` relies on a static link-time coupling via `ETH_IRQHandler` to its target startup vector table, rather than using `NVIC_SetVector` to set it at runtime.

Could your more complex environment have somehow lost this link, or otherwise masked the interrupt?

Although, actually, the link status callback "netif_link_irq" is not via the Ethernet IRQ, but via polling of the PHY in `STM32_EMAC::phy_task`. If you're not getting that callback, something worse is happening.

Is there a more fundamental OS/environment problem here? Is phy_task not running due to some other thread starving it? Has the data section not been correctly initialised?

UsmanKhan0392, 2018-11-01:
Basically my code had some linking issue due to which the it didn't enter `HAL_ETH_MspInit` which was overridden by `stm32f4_eth_init.c` for my target.

Rather it entered it's weaker implementation. 

Due to this `ETH_HandleTypeDef` wasn't initialized properly and therefore `HAL_ETH_WritePHYRegister `couldn't write to it's registers.

Now that I fixed the code to enter the overridden function, it works fine.

closed 2018-11-01 by UsmanKhan0392