ARMmbed/mbed-os #8494: SDBlockDevice error

pilotak, 2018-10-22:
### Description
When using SDBlockDevice from external repo i get an error, which is fine
`[Error] SDBlockDevice.cpp@157,2: #error "Started from version 5.10.0 SDBlockDevice external repo is deprecated. please use the SDBlockDevice component inside mbed-os."`

so i deleted the folder and i get `fatal error: SDBlockDevice.h: No such file or directory`

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

linlingao, 2018-10-22:
It looks like you might not have upgraded your Mbed OS to 5.10. 
You should see SDBlockDevice.h in the following directory:
<your_project_dir>/mbed-os/components/storage/blockdevice/COMPONENT_SD/SDBlockDevice.h.
If you don't see this file or even the directory, please upgrade your Mbed OS to 5.10.

pilotak, 2018-10-23:
@linlingao yes i have the latest OS.

Thank you @yennster for pointing me to the right direction however it is not very clear in documentation for devices that do not have `BlockDevice` on development board. Anyway here is how i got it to work if somebody else struggles.

`mbed_app.json`
```json
{
    "target_overrides": {
        "NUCLEO_F412ZG": {
            "target.features_add": ["STORAGE"],
            "target.components_add": ["SD"]
        }
    }
}
```

linlingao, 2018-10-23:
Reopen issue for doc enhancement.

closed 2019-06-05 by ciarmcom

linlingao, 2019-06-05:
Fixed by https://github.com/ARMmbed/sd-driver/pull/115.

wdx04, 2023-06-03:
For anyone who want to use SD cards with bare metal profile : it is also needed to add "storage" and "sd" to the "requires" section:
"requires": ["bare-metal", "events", "storage", "sd", "filesystem", "fat_chan"]