ARMmbed/mbed-os #8500: LoRaWAN: #8405 Considerably Increases LoRaWAN Stack Size

mattbrown015, 2018-10-23:
### Description
@hasnainvirk

I'm trying to get our mbed-os up-to-date because there are some fixes we need.  _But_, every step I take causes a problem!

I've just tried to move to [#adf9165](https://github.com/ARMmbed/mbed-os/commit/adf9165af2d1ef04e87c1022a71d05f6cafa3d6d) which is the version where #8405 is merged.  This failed to link because I ran out of flash! :-(

I created a very simple test app and can see that moving to #8405 increases the code size by 832 bytes.  This seems like a big increase in relation to the small devices we're using.

    | Module                                                                                                      |        .text |    .data |       .bss |
    |-------------------------------------------------------------------------------------------------------------|--------------|----------|------------|
    ...
    | mbed-os\features\lorawan\LoRaWANInterface.o                                                                 |     1345(+0) |    0(+0) |      0(+0) |
    | mbed-os\features\lorawan\LoRaWANStack.o                                                                     |   6461(+492) |    0(+0) |      0(+0) |
    | mbed-os\features\lorawan\lorastack\mac\LoRaMac.o                                                            |   7170(+305) |    0(+0) |      0(+0) |
    | mbed-os\features\lorawan\lorastack\mac\LoRaMacChannelPlan.o                                                 |      408(+0) |    0(+0) |      0(+0) |
    | mbed-os\features\lorawan\lorastack\mac\LoRaMacCommand.o                                                     |     1486(+0) |    0(+0) |      0(+0) |
    | mbed-os\features\lorawan\lorastack\mac\LoRaMacCrypto.o                                                      |      844(+0) |    0(+0) |      0(+0) |
    | mbed-os\features\lorawan\lorastack\phy\LoRaPHY.o                                                            |    4306(+20) |    0(+0) |      0(+0) |
    | mbed-os\features\lorawan\lorastack\phy\LoRaPHYEU868.o                                                       |      802(+0) |    0(+0) |      0(+0) |
    | mbed-os\features\lorawan\system\LoRaWANTimer.o                                                              |      253(+0) |    0(+0) |      0(+0) |
    ...
    | Subtotals                                                                                                   | 113816(+832) | 2944(+0) | 11616(+24) |
    Total Static RAM memory (data + bss): 14560(+24) bytes
    Total Flash memory (text + data): 116760(+832) bytes

I haven't looked in any detail at #8405 but there seem to be a number of things going on.

We're a class A device and therefore, if the size change is related to class C code it's a waste.

Is there anything that can be done to reduce the impact of #8405?

I've been thinking for a while that the LoRaWAN stack and the radio drivers would benefit from some configuration.  Not including support for class C in a class A device is one example.  Another example in the radio drivers, would be removing support for FSK when only Lora is needed.

### Issue request type
    [X] Question
    [ ] Enhancement
    [ ] Bug

mattbrown015, 2018-10-23:
I've manged to get our mbed-os updated to the _latest_ by removing/shortening our debug trace strings.

I can see the changes in LoraMac.cpp and LoRaWANStack.cpp.  Unfortunately it all looks like _genuine_ code (whatever that means!).

So there's no immediate problem but it's a bit of a worry that things like this can happen.

0xc0170, 2018-10-23:
> I created a very simple test app and can see that moving to #8405 increases the code size by 832 bytes. This seems like a big increase in relation to the small devices we're using.

@ARMmbed/mbed-os-wan Please review

hasnainvirk, 2018-10-24:
@mattbrown015 Sad to hear that you were having issues. Sorry for late reponse, we are in Tokyo for LoRa Alliance meetings. How much flash do you have onboard ? 128K should be more than enough for simpler applications. Which toolchain do you use to build ? In production, we should use release build rather than develop or debug. Please let me know what is your target platform ? 
#8405 is mostly related to QOS which is valid for all classes including Class A. 
I am on vacation from tomorrow till 13th of Nov :) however some other team member would surely be looing into this. Just for sake of information, Can you afford going non-RTOS ?
I will try my best to look for any savings we can make in this regard, If I get details about your toolchain, target platform and build type until tomorrow.

mattbrown015, 2018-10-24:
Hitting another problem as I tried to update mbed-os made me a bit grumpy but I've cheered up now that I'm back on track! :-)

I think my initial assessment was a bit off.  `6461(+492)` is an 8% increase and `7170(+305)` is a 4% increase; so not too bad and if the work was needed 'it was needed'.

We're using an STM32L4 with a whopping 256K but 'simple' is a subjective word.  My project manager thinks what we're trying to do is 'simple' and tells me so every time he asks why we're not finished yet! :-}

We regularly discuss moving to release and/or removing the debug trace but at present we still don't have the confidence and it's not clear when that confidence will come.

I think might point about the radio drivers might have more mileage.

hasnainvirk, 2018-10-24:
Release build 102K, Developer build 114K, Debug build -> 179K.
These are the numbers I am getting GCC for flash. I assume your application is much heavier than our little example here so I can imagine soon you will hit the wall. 
I was reading your comments on another issue and I would really encourage you to consider a little redesign of your app, before it gets way out of hand :). What I mean is that we shouldn't use Mbed assert and mbed die for application logic. If an assert was hit, it would have been a pure, legitimate mistake which should not have occured. If for example the stack hits an assert, that means we didn't test the stack enough and we need to improve our tests. A better solution for you may be to save your session or any important information into NVRAM and just power cycle the device in emergency cases.

mattbrown015, 2018-10-24:
I think we're getting off the point now but...

We're not using `MBED_ASSERT` or `mbed_die` for application logic.  We're using `MBED_ASSERT` to catch mistakes and misunderstandings by the programmer.  It's not our intention to ever fire an assert.

hasnainvirk, 2018-10-25:
You can reduce the size of your build by changing the -O0 flag in your mbed-os/tools/profiles/debug.json to -Os for your debug builds. Please note that the I am not sure what side effects this change will bring forth as I can only compile an app at the moment. 
The number that I came up  with is 115K with GCC debug build. Please test if that will do the trick for you. Keep me posted please.

hasnainvirk, 2018-10-25:
@mattbrown015 I was just talking to one of our OS experts and I heard that you could use MBED_ASSERT with release build by just changing a couple of lines manually. Would you like to hear more about it ?

mattbrown015, 2018-10-25:
@hasnainvirk Thanks for  thinking about me!  I'm sorry I think I've distracted you much more than I intended or was necessary.

I appreciate that there are lots options open to us but, at the moment, we're back on track and I'm reasonably happy with what we've done.  Nothing is ever perfect and there are always things that could/should be done differently!

Most of our features are complete now so I'm hoping that we're heading for a period of stability that will allow us to get some soak testing done.

I think mbed-os and the mbed-os community has worked pretty well for us. 💯

closed 2018-10-25 by mattbrown015

0xc0170, 2018-10-25:
>I think mbed-os and the mbed-os community has worked pretty well for us. 💯

Thanks! 🎉 👍