ARMmbed/mbed-os #8501: Setting the MAC address before connection on atmel radio is not possible.

javier-moreno-tridonic-com, 2018-10-23:
### Description

We are trying to migrate our codebase from mbedOS 5.7 to 5.10. We have a range of MAC addresses available for our device. The MAC address is pre-flashed in section of our rom after the device is produced. So we do not use any MAC chip. We provide the MAC address ourselves.

On mbed 5.7 we could achieve this by calling the virtual method **NanostackPhy::set_mac_address** that was implemented in **NanostackRfPhyAtmel::set_mac_address**.

The problem comes because on 5.7 **platform_critical_init()** was called when **ThreadInterface::initialize(NanostackRfPhy \*phy)** was called. 

[image]

This allowed the Mutex used by **rf_if_lock();** to be initiliazed. So we could do the following

- Initialize our interface with **ThreadInterface::initialize(NanostackRfPhy \*phy)**
- Set the mac address with **NanostackPhy::set_mac_address**
- And finally do a **ThreadInterface::connect()**

With the current structure in place the call to **mesh_system_init();** (which will eventualy lead to the call of **platform_critical_init()**) was moved to the constructor of  [Nanostack::Interface::Interface](https://github.com/ARMmbed/mbed-os/blob/8054060bf289624be6d39c6e96d709f976168d91/features/nanostack/mbed-mesh-api/source/MeshInterfaceNanostack.cpp#L74) 

As far as I could trace, with the current code, the constructor will only get called when we do **ThreadInterface::connect()**

[image]

This means that we can't provide our MAC the way we used to, as doing so will trigger the following error because the Mutex is uninitialized.

```
++ MbedOS Error Info ++<\n>
Error Status: 0x80010133 Code: 307 Module: 1<\n>
Error Message: Mutex: 0x0, Parameter error<\n>
Location: 0x8038DD7<\n>
Error Value: 0x0<\n>
Current Thread: Id: 0x20014BFC Entry: 0x8038AFD StackSize: 0x1800 StackMem: 0x200133A8 SP: 0x2003FF58 <\n>
-- MbedOS Error Info --
```

IMHO there should be the possibility to set the mac externally without needing to rely in the specific implementation of the atmel-rf-driver (which I presume right now it just adapted to the arduino shields that come with a MAC chip integrated)

Is there anything I'm missing that would allow me to do what we intend?

### Issue request type

    [ ] Question
    [X] Enhancement
    [X] Bug

CC @MarceloSalazar

markus-becker-tridonic-com, 2018-10-23:
That would work, but it seems it is part of private API, or?

adbridge, 2018-10-23:
@javier-moreno-tridonic-com Looks like you got the above comment because you ticked multiple boxes in the PR type

javier-moreno-tridonic-com, 2018-10-24:
@adbridge Done

@kjbracey-arm I tried your proposed solution and it works. Is there any plans to make that fix permanent?

kjbracey-arm, 2018-10-24:
> That would work, but it seems it is part of private API, or?

No, it's fine. It's effectively a centralised call to initialise the nanostack HAL components. A native Nanostack application would have that start-up sequence in a single place at the start of main. When we have multiple bits of code using the Nanostack HAL, and they need to work transparently to non-Nanostack-aware mbed OS apps, each of them needs to ensure the HAL is active before they use it. `mesh_system_init` is what they all call, and it ensures initialisation happens on first call only.

Could you submit it as a PR?

javier-moreno-tridonic-com, 2018-10-24:
> Could you submit it as a PR?

Currently the code base I work on, is actually a fork of mbed-os with more modifications we need for our products. So is not a straightforward thing to do for me at the moment, but I could try at some point in time as in the end, the modification is a couple of one-liners. But that would take me to the next issue: wouldn't moving this break the implementation for other radios? If that's the case, we wouldn't have the possibility to test the PR properly.

kjbracey-arm, 2018-10-24:
Other radios might need the equivalent fix, but adding this won't break anything.

> moving this

Ah, I wasn't suggesting moving the `mesh_system_init` - you should be adding an extra call. First call to `mesh_system_init` does the work - makes it startup order independent.

javier-moreno-tridonic-com, 2018-10-24:
@kjbracey-arm I see. I misunderstood/misread your suggestion. Then it should truly be a one-liner :)

I'll try that on my codebase (just to be sure and open a the PR if it works)

markus-becker-tridonic-com, 2018-11-06:
@javier-moreno-tridonic-com I think this solution works for us. If you agree, please close the issue.

javier-moreno-tridonic-com, 2019-03-13:
Finally changes were done in our atmel-rf-driver fork as the suggested location in the constructor would affect other of our targets. I guess that a call at the beginning of main would also work so I don't see a reason anymore to propose a modification via PR anymore

closed 2019-03-13 by javier-moreno-tridonic-com