ARMmbed/mbed-os #8513: NVSTORE deinit function doesn't free memory

jeromecoutant, 2018-10-23:
### Description

Got this test failed during ST CI tests :

Target | Toolchain | Test_ID/TestSuite | Test_Description/TestCase | Status
-- | -- | -- | -- | --
NUCLEO_F070RB | ARM | features-storage-nvstore-tests-nvstore-functionality | NVStore: Multiple   thread test | ERROR
NUCLEO_F072RB | ARM | features-storage-nvstore-tests-nvstore-functionality | NVStore: Multiple   thread test | ERROR

Console log:

````
[1540096136.96][CONN][RXD] >>> Running case #3: 'NVStore: Multiple thread test'...
[1540096137.01][CONN][INF] found KV pair in stream: {{__testcase_start;NVStore: Multiple thread test}}, queued...
[1540096144.03][CONN][RXD]
[1540096144.03][CONN][RXD]
[1540096144.06][CONN][RXD] ++ MbedOS Error Info ++
[1540096144.11][CONN][RXD] Error Status: 0x8001011F Code: 287 Module: 1
[1540096144.15][CONN][RXD] Error Message: Operator new[] out of memory
[1540096144.15][CONN][RXD]
[1540096144.17][CONN][RXD] Location: 0x800A5CB
[1540096144.20][CONN][RXD] Error Value: 0x104
[1540096144.29][CONN][RXD] Current Thread: Id: 0x20001F20 Entry: 0x800BB91 StackSize: 0xC00 StackMem: 0x20001320 SP: 0x20001CE0
[1540096144.39][CONN][RXD] For more info, visit: https://armmbed.github.io/mbedos-error/?error=0x8001011F
[1540096144.40][CONN][RXD] -- MbedOS Error Info --
````

### Analysis

It seems that issue comes there :

    ret = nvstore.deinit();
    TEST_ASSERT_EQUAL(NVSTORE_SUCCESS, ret);
    ret = nvstore.init(); ==> Operator new[] out of memory

It seems that the **deinit** function doesn't deallocate the memory,
as in the next **init**, some memory is missing ?

@geky @davidsaada 

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

davidsaada, 2018-12-06:
@jeromecoutant Working on it now. Seems like `deinit` is not the problem, but will need to make sure.

davidsaada, 2019-01-07:
@jeromecoutant Is this reproduced?

davidsaada, 2019-01-21:
Thanks @jeromecoutant managed to reproduce it (by simulating a memory restricted board). Trying to find the root cause now.

davidsaada, 2019-01-21:
@jeromecoutant Please check whether #9445 resolves this issue.

PR #9445 (2019-01-31): NVStore: fix area calculation function
### Description
NVStore includes a function (`calc_validate_area_params`) that either calculates or validates the addresses and sizes of both areas. This function allocates a sector map array used in both calculation/validation of areas. While this generally worked, it had a few problems: First of all, it allocated and deallocated the sector map inside the function, while a more efficient code wouldn't need it. Second, it had a potential bug - number of of entries in the array was calculated according to the first sector size, but in case of a variant sec [... cut]
closed 2019-01-24 by ciarmcom