ARMmbed/mbed-os #8520: UBLOX_EVK_ODIN_W2 does not pass Greentea testcase WIFI-CONNECT

VeijoPesonen, 2018-10-24:
### Description
UBLOX_EVK_ODIN_W2 does not pass Greentea testcase WIFI-CONNECT.

**Target**
UBLOX_EVK_ODIN_W2

**Toolchain:**
GCC_ARM, ARM, IAR

**mbed-os sha:**
`f9a723e`

**Expected behavior**
WIFI-CONNECT succeeds

**Actual behavior**
Third connection attempt on the test case will fail due to the reason that driver tries to read SSID from a memory location not managed by the driver and it has been trashed. Assumption is that driver would keep the credentials (SSID, password) stored between different connection attempts.

```
[1540361853.17][CONN][RXD] >>> Running case #11: 'WIFI-CONNECT'...
[1540361853.18][CONN][INF] found KV pair in stream: {{__testcase_start;WIFI-CONNECT}}, queued...
[1540361857.20][CONN][RXD] :47::FAIL: Expected 0 Was -3003
[1540361857.29][CONN][RXD] :48::FAIL: Expected 0 Was -3004
[1540361857.29][CONN][INF] found KV pair in stream: {{__testcase_finish;WIFI-CONNECT;0;2}}, queued...
[1540361857.40][CONN][RXD] >>> 'WIFI-CONNECT': 0 passed, 2 failed with reason 'Test Cases Failed'

```

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

VeijoPesonen, 2018-10-24:
@andreaslarssonublox please check.

andreaslarssonublox, 2018-10-24:
Handing it over to @aqib-ublox

asifrizwanubx, 2018-10-24:
@ARMmbed/team-ublox taking team in loop.

screamerbg, 2018-10-24:
@VeijoPesonen Please use team tag instead @ARMmbed/team-ublox

AmmadRehmat, 2018-10-25:
Looking into it.
Internal jira reference https://jira.u-blox.net/browse/TE_OM2-250

AmmadRehmat, 2018-10-29:
@VeijoPesonen  PR has been generated to fix this issue.

https://github.com/ARMmbed/mbed-os/pull/8566

closed 2018-12-12 by VeijoPesonen