ARMmbed/mbed-os #8523: Enter DFU mode with stm32F401 (bootloader Issue)

devHAA, 2018-10-24:
Hello,

I'm having troubles with mbed-os 5. With the previous version of mbed (v163) it worked perfectly. With the following piece of assembler code, the micro (stm32F401) enters DFU mode. But with the newest version of mbed-os 5 doesn't work.

```c++
void enterDFU(){
 
     __asm ("LDR     R0, =0x40023844");// RCC_APB2ENR
     __asm ("LDR     R1, =0x00004000");// ENABLE SYSCFG CLOCK
     __asm ("STR     R1, [R0, #0]");
 
     __asm ("LDR     R0, =0x40013800");// remap ROM at zero
     __asm ("LDR     R1, =0x00000001");// SYSCFG_MEMRMP
     __asm ("STR     R1, [R0, #0]");
 
     __asm ("LDR     R0, =0x1FFF0000");// ROM BASE
     __asm ("LDR     SP,[R0, #0]");    // SP @ +0
     __asm ("LDR     R0,[R0, #4]");    // PC @ +4
 
     __asm ("BX      R0");             // Jump to Bootloader
}
```
I've been doing some tests and I think that the problem is in the last line  `__asm ("BX      R0");` , when it should jump to the Bootloader. If I try to connect the device on the PC it says "descriptor request error".

Any ideas what may be happening, Thank you.

0xc0170, 2018-10-24:
Just to be certain - you are using f401 with v163 (mbed 2 version), not Mbed OS 5.10.2 ?

@ARMmbed/team-st-mcd Please review

devHAA, 2018-10-25:
> Can you compare R0 value before and now? What toolchain - using online compiler?

I'm using eclipse with GCC_ARM compiler. In the online compiler the `__asm `is not recognized. The compiler error says "_Expected an expression "__asm ("LDR R0, =0x40023844");_"

I'm trying to see how to read R0 in c++, when I know how to do it I'll post it here
 
> Look at ST forum to find some help - missing driver for instance?

This is not the problem, because as I said, with mbed 2 (v163) it works, and the PC takes the proper driver because the driver is installed.

devHAA, 2018-10-25:
> OK, might be the board is in bad state and cant connect.

I just tried this simple example with the **same board** (NUCLEO-F401RE), using mbed 2 (v163), and then mbed-os 5.10.2. Both with the online compiler.

```
#include "mbed.h"

#if defined(__CC_ARM)
    __asm static void enterDFU(){
        LDR     R0, =0x40023844
        LDR     R1, =0x00004000
        STR     R1, [R0, #0]
        
        LDR     R0, =0x40013800
        LDR     R1, =0x00000001
        STR     R1, [R0, #0]
        
        LDR     R0, =0x1FFF0000
        LDR     SP,[R0, #0]
        LDR     R0,[R0, #4]
        BX      R0
    }
#endif

int main() {     
    enterDFU();  
}
```
With the mbed 2 (v163) it works perfect (PC recognized it as **_STM Device in DFU Mode_**), but with mbed-os 5.10.2 doesn't work. It says _**Unknown USB Device (Device Descriptor Request Failed)**_

elm3, 2019-01-07:
is the ST team (@ARMmbed/team-st-mcd) reviewing or aware?

devHAA, 2019-01-24:
@ciarmcom @adbridge @0xc0170  any updates with this issue?

I'm stuck with mbed2 because I can't use mbed-os without the DFU mode working.

Please, can anyone from the ARMmbed team tell us if this is being investigated?

Thank you.

cmonr, 2019-01-30:
@devHAA Right now, it seems like this is waiting on @ARMmbed/team-st-mcd to take a look.

jeromecoutant, 2019-01-31:
- v5.9.6: OK
- v5.10.2: FAILED

Maybe you can check intermediate version ?

devHAA, 2019-02-05:
> * v5.9.6: OK
> * v5.10.2: FAILED
> 
> Maybe you can check intermediate version ?

Ok, after some test, still no lucky:

v5.9.6 -> failed
v5.10.2 -> failed
v5.11.3 -> failed (latest release)

jeromecoutant, 2019-02-05:
I would like to know what happened between 5.9.6 and 5.10.2 ?
Could you check 5.9.7, 5.10.0 and 1 ....?

devHAA, 2019-02-05:
> I would like to know what happened between 5.9.6 and 5.10.2 ?
> Could you check 5.9.7, 5.10.0 and 1 ....?

Just checked v5.9.6 but is not working for me neither, is it working for you?

neurodigitaltechnologies, 2019-06-11:
Dear all,
How is this bug going on, please? I think we are a lot of people stuck with MBED OS 2 because of this.
@devHAA @jeromecoutant @ciarmcom @adbridge @0xc0170 
Is there any solution, fix or workaround?
Can you elevate the priority, please? We have an upgrade in the technology roadmap pending on this issue to make a final decision about next platform.
Many thanks in advance!

fpistm, 2019-06-11:
Hi @devHAA and all,

could you test this:
```
#include "mbed.h"

#if defined(__CC_ARM)
    __asm static void enterDFU(){
        LDR     R0, =0x40023844 // RCC_APB2ENR
        LDR     R1, =0x00004000 // ENABLE SYSCFG CLOCK
        STR     R1, [R0, #0]
        
        LDR     R0, =0x40013800 // remap ROM at zero
        LDR     R1, =0x00000001 // SYSCFG_MEMRMP
        STR     R1, [R0, #0]
        
        LDR     R0, =0x1FFF0000 // ROM BASE
        LDR     SP,[R0, #0]     // SP @ +0
        LDR     R0,[R0, #4]     // PC @ +4
        BX      R0              // Jump to Bootloader
    }
#endif

gpio_t objDP;

int main() {
    gpio_init(&objDP, PA_12);
    gpio_dir(&objDP, PIN_OUTPUT);
    gpio_write(&objDP, 0);
    wait(0.01);
    gpio_init(&objDP, PA_12);
   
    enterDFU();  
}
```

jeromecoutant, 2019-06-17:
@neurodigitaltechnologies 
Any feedback ?
Thx

jjjuande, 2019-08-12:
> @neurodigitaltechnologies
> Any feedback ?
> Thx

Hi.

I managed to get this code working on a STM32F401 with mbed-os 5 (updated with the latest commit f8dc035ae4221626423cf99a4a216868cd599393 from Aug 9, 2019) only under special conditions and in a consistent way. In fact, all the other code posted in this thread works under the same circumstances too. I'll post my experience in case someone can explain it.

```c
#include "mbed.h"

DigitalOut led1(PC_9);  // I need the led to help me know when it's running

void enterDFU(){
	// SYSCFG_MEMRMP / remap ROM at zero
	 __asm ("LDR     R0, =0x40013800 ; ");
	 __asm ("LDR     R1, =0x00000001 ; ");
	 __asm ("STR     R1, [R0, #0]    ;");
	 // Jump to Bootloader
	 __asm ("LDR     R0, =0x1FFF0000 ; ");
	 __asm ("LDR     SP,[R0, #0]     ; ");
	 __asm ("LDR     R0,[R0, #4]     ; ");
	 __asm ("BX      R0");
}

int main()
{
	enterDFU();
}
```

If I turn on the STM32F401 while connected to my PC by USB, I get the error: "Unknown USB Device (Device Descriptor Request Failed)". 

if I turn on the STM32F401 while NOT connected to my PC, and then I connect the USB to my PC, I get the same problem. 

The only way for this to work is to connect the STM32F401 to the ST-LINK/V2 (not connected yet to the PC) using SWD (in my case), turn on the STM32F401 (so the enterDFU code runs first thing) and then connect the ST-LINK/V2 to the PC using the USB cable. Now, if you connect the STM32F401 to the PC, you get the "STM Device in DFU Mode".

After that, I can unplug the ST-LINK/V2 USB cable from the PC, plug/unplug the STM32F401 from the PC and it works all the time while the SWD connection is kept (to the ST-LINK/V2 which isn't connected to the PC). Turning the STM32F401 on/off works too. The first time it fails, but after that, it works all the time.

Can someone help me to understand this?

Thanks.

neurodigitaltechnologies, 2019-08-12:
Hi.

I managed to get this code working on a STM32F401 with mbed-os 5 (updated with the latest commit f8dc035 from Aug 9, 2019) only under special conditions and in a consistent way. In fact, all the other code posted in this thread works under the same circumstances too. I'll post my experience in case someone can explain it.

#include "mbed.h"
DigitalOut led1(PC_9);  // I need the led to help me know when it's running

void enterDFU(){
	// SYSCFG_MEMRMP / remap ROM at zero
	 __asm ("LDR     R0, =0x40013800 ; ");
	 __asm ("LDR     R1, =0x00000001 ; ");
	 __asm ("STR     R1, [R0, #0]    ;");
	 // Jump to Bootloader
	 __asm ("LDR     R0, =0x1FFF0000 ; ");
	 __asm ("LDR     SP,[R0, #0]     ; ");
	 __asm ("LDR     R0,[R0, #4]     ; ");
	 __asm ("BX      R0");
}

int main()
{
	enterDFU();
}

If I turn on the STM32F401 while connected to my PC by USB, I get the error: "Unknown USB Device (Device Descriptor Request Failed)".

if I turn on the STM32F401 while NOT connected to my PC, and then I connect the USB to my PC, I get the same problem.

The only way for this to work is to connect the STM32F401 to the ST-LINK/V2 (not connected yet to the PC) using SWD (in my case), turn on the STM32F401 (so the enterDFU code runs first thing) and then connect the ST-LINK/V2 to the PC using the USB cable. Now, if you connect the STM32F401 to the PC, you get the "STM Device in DFU Mode".

After that, I can unplug the ST-LINK/V2 USB cable from the PC, plug/unplug the STM32F401 from the PC and it works all the time while the SWD connection is kept (to the ST-LINK/V2 which isn't connected to the PC). Turning the STM32F401 on/off works too. The first time it fails, but after that, it works all the time.
Can someone help me to understand this?
Thanks.

closed 2019-10-11 by adbridge