ARMmbed/mbed-os #8527: Invalid type conversion in "extras/mbed-os.lib/features/netsocket/NetworkStack.h"

willfurnell, 2018-10-24:
### Description

It's quite possible this isn't an issue with mbed-os as I'm quite new to using it! Nevertheless...

When trying to compile the following program:

https://os.mbed.com/users/co657_wf44/code/net/

in the online compiler I get the following error:

Error: Invalid type conversion in "extras/mbed-os.lib/features/netsocket/NetworkStack.h", Line: 419, Col: 32

I've narrowed it down to trying to send data over the UDP socket - in the minium example I'm trying to just send a single char over the socket and the compilation error still persists.

I'm using mbed-os as of commit https://github.com/ARMmbed/mbed-os/commit/5b25b6643d24c94442695a858fc9a5845d5c0957

I'm compiling for the FRDM K64F board.

Thank you,

Will.

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

kjbracey-arm, 2018-10-25:
Socket constructor needs a pointer to a `NetworkStack` or `NetworkInterface`, which is what `net` is. Code should read:

    UDPSocket socket(net);

willfurnell, 2018-10-25:
Thank you very much for your help!

Will.

0xc0170, 2018-10-25:
@willfurnell If this answers your question, please close the issue

closed 2018-10-25 by willfurnell