ARMmbed/mbed-os #8544: Merging bootloader fails using Python3

GarySwansonARM, 2018-10-25:
### Description
(Note: this is not a problem with Python 2.7.14)

On a mac I followed the following steps
- create a new virtualenv with python 3.7.0:
virtualenv -p /usr/local/bin/python3 test3
- cd test3
- run "source bin/activate"
- pip install mbed-cli
- mbed-cli --version -> 1.8.2
- mbed import https://os.mbed.com/teams/ST/code/example-DISCO_L475_IOT-mbed-Cloud-connec/
- cd example-DISCO_L475_IOT-mbed-Cloud-connec 
- get the developer certificates by using mbed dm init:
mbed dm init -d "Acme.com" --model-name "model1" -a <application key> --force
- mbed compile -t GCC_ARM -m DISCO_L475VG_IOT01A -vv

the error I got was:
```
Link: /usr/local/bin/gcc-arm-none-eabi-6-2017-q2-update/bin/arm-none-eabi-gcc @./BUILD/DISCO_L475VG_IOT01A/GCC_ARM/.link_options.txt
[DEBUG] Return: 0
Elf2Bin: example-DISCO_L475_IOT-mbed-Cloud-connec_application
FromELF: /usr/local/bin/gcc-arm-none-eabi-6-2017-q2-update/bin/arm-none-eabi-objcopy -O binary ./BUILD/DISCO_L475VG_IOT01A/GCC_ARM/example-DISCO_L475_IOT-mbed-Cloud-connec_application.elf ./BUILD/DISCO_L475VG_IOT01A/GCC_ARM/example-DISCO_L475_IOT-mbed-Cloud-connec_application.bin
[DEBUG] Return: 0
Merging Regions
  Filling region bootloader with /Users/garswa01/test3/example-DISCO_L475_IOT-mbed-Cloud-connec/bootloader/mbed-bootloader-L475VG.bin
  Padding region bootloader with 0x14 bytes
Traceback (most recent call last):
  File "/Users/garswa01/test3/example-DISCO_L475_IOT-mbed-Cloud-connec/mbed-os/tools/make.py", line 293, in <module>
    ignore=options.ignore
  File "/Users/garswa01/test3/example-DISCO_L475_IOT-mbed-Cloud-connec/mbed-os/tools/build_api.py", line 548, in build_project
    merge_region_list(region_list, res, notify)
  File "/Users/garswa01/test3/example-DISCO_L475_IOT-mbed-Cloud-connec/mbed-os/tools/build_api.py", line 423, in merge_region_list
    _fill_header(region_list, region).tofile(header_filename, format='hex')
  File "/Users/garswa01/test3/example-DISCO_L475_IOT-mbed-Cloud-connec/mbed-os/tools/build_api.py", line 380, in _fill_header
    header.puts(start, struct.pack(fmt, time()))
struct.error: required argument is not an integer
[mbed-64219] ERROR: "/Users/garswa01/test3/bin/python3.7" returned error.
       Code: 1
       Path: "/Users/garswa01/test3/example-DISCO_L475_IOT-mbed-Cloud-connec"
       Command: "/Users/garswa01/test3/bin/python3.7 -u /Users/garswa01/test3/example-DISCO_L475_IOT-mbed-Cloud-connec/mbed-os/tools/make.py -t GCC_ARM -m DISCO_L475VG_IOT01A --source . --build ./BUILD/DISCO_L475VG_IOT01A/GCC_ARM -v"
```

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

0xc0170, 2018-10-25:
What's your mbed OS version (latest release? previous one? tested also on master) ?

GarySwansonARM, 2018-10-25:
> What's your mbed OS version (latest release? previous one? tested also on master) ?

I tried 5.10.1 (comes with the example app) and 5.10.2

MarceloSalazar, 2018-11-01:
Thanks for letting us know! FYI @cmonr @OPpuolitaival

kaidokert, 2018-11-16:
changing 
> header.puts(start, struct.pack(fmt, time()))

to 
> header.puts(start, struct.pack(fmt, int(time())))

makes it work. Is a PR ok for this ?

kaidokert, 2018-11-18:
There is one more bug on this line: 

https://github.com/ARMmbed/mbed-os/blob/master/tools/build_api.py#L391

Has `fmt = {"CRCITT32be": ">l", "CRCITT32le": "<l"}[subtype]`
Should be `fmt = {"CRCITT32be": ">L", "CRCITT32le": "<L"}[subtype]`

Currently throws errors randomly depending on the computed CRC value:
```
    header.puts(start, struct.pack(fmt, zlib.crc32(ih.tobinarray())))
struct.error: 'l' format requires -2147483648 <= number <= 2147483647
```

cmonr, 2018-12-04:
@GarySwansonARM @kaidokert Is this sitll an issue?

I recall seeing PR activity around the build_api.py file recently.

0xc0170, 2018-12-12:
As the fix was reverted (breakage caused), not sure if this was addressed by any other. @GarySwansonARM Please can you retest

GarySwansonARM, 2018-12-12:
I just tried this with a fresh install of mbed cli 1.8.3 and it is still not fixed.

closed 2019-02-05 by ciarmcom

cmonr, 2019-02-11:
@SenRamakri Remember that comments made on the Jira issue are private by default, and are not sync'd to GitHub.

This was closed as a duplicate of https://github.com/ARMmbed/mbed-os/issues/8146