ARMmbed/mbed-os #8555: NVSTORE test issue

jeromecoutant, 2018-10-26:
### Description

Hi

features-storage-nvstore-tests-nvstore-functionality is failed for NUCLEO_F410RB.

Issue is the same as in #7212
Fix seems to be not sufficient.
I checked my validation status, but it seems that I checked only flashiap tests at that time.
I forgot nvstore test...
Sorry to reopen this point.

@davidsaada 

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

jeromecoutant, 2018-10-26:
https://github.com/ARMmbed/mbed-os/blob/fc741f03a1b6a46b80a54d8bae723427955d1662/features/storage/nvstore/source/nvstore.h#L69

I think the problem is that NVSTORE needs 2 FLASH area.
As explained in #7149, due to the memory mapping, some targets can't used 2 areas.

0xc0170, 2018-10-26:
@ARMmbed/mbed-os-storage

dannybenor, 2018-11-20:
This cannot be fixed. NVStore requires 2 sectors

dannybenor, 2018-11-20:
I think this can be closed

jeromecoutant, 2018-11-20:
So how can I undefine NVSTORE for NUCLEO_F410RB ?

dannybenor, 2018-11-20:
Yes

jeromecoutant, 2018-11-26:
No...
NUCLEO_F410RB supports FLASH!
It doesn't support NVSTORE

dannybenor, 2018-11-26:
OK
You can use https://github.com/ARMmbed/mbed-os/blob/master/features/storage/nvstore/mbed_lib.json to disable NVStore for the relevant targets.
Just for my curiosity, what will be the effect of removing "FLASH" for this target?

PR #8951 (2018-12-04): Remove NVSTORE enabled from NUCLEO_F410RB
### Description

This fixes #8555 

NVSTORE needs 2 FLASH area, which is not available for NUCLEO_F410RB

### Pull request type

    [X] Fix
    [ ] Refactor
    [ ] Target update
    [ ] Functionality change
    [ ] Docs update
    [ ] Test update
    [ ] Breaking change
closed 2018-12-04 by cmonr