ARMmbed/mbed-os #8558: Consistent behavior of us_ticker_init accross targets

LaurentLouf, 2018-10-26:
### Description

- Target STM
- GNU ARM 7-2017-q4-major
- mbed-cli 1.7.3
- 603c4f9 (HEAD -> master) Merge pull request #7851 from geky/fix-littlefs-mount-cleanup

I have noticed that compared to most of other targets, the `us_ticker_init` on STM targets does not any mechanism to check if the ticker has already been initialized. For example, on target MAX_32620, we have the following, simple yet effective : 
```
void us_ticker_init(void)
{
    if (us_ticker_inited)
        return;
    us_ticker_inited = 1;
    [...]
}
```
And we have something similar for a lot of other targets, but at least the target STM is missing this feature. This would be nice to have a consistent behavior across all targets to be sure that calling multiple times `us_ticker_init` will only initialize the ticker once.

### Issue request type

    [ ] Question
    [X] Enhancement
    [ ] Bug

0xc0170, 2018-10-26:
From the specs:

>  * * The function ticker_init is safe to call repeatedly - Verified by test ::ticker_init_test
>  * * The function ticker_init allows the ticker to keep counting and disables the ticker interrupt - Verified by test ::ticker_init_test

@c1728p9

c1728p9, 2018-10-26:
Hi @LaurentLouf the ticker specification requires that calling `ticker_init` disables ticker interrupts. This is to ensure that when using an application with a bootloader, any pending ticker events setup by the bootloader are not unintentionally triggered in the application during `ticker_init`.

The code you have show above doesn't conform to this behavior so this is actually a bug in the MAX32620. All devices should perform some initialization on multiple calls to `us_ticker_init`.

One part of `us_ticker_init` that is implementation defined is if the internal counter is reset or keeps counting where it left off. The MCU vendor can change this behavior to best suite the hardware. The mbed code making use of the ticker uses an offset from when `us_ticker_init` was called so it works the same in both cases.

elm3, 2019-01-07:
@ARMmbed/team-st-mcd

closed 2019-10-31 by ciarmcom