ARMmbed/mbed-os #8568: examples.py should fail when network issues are encountered.

cmonr, 2018-10-29:
### Description

Currently, the build wrapper job does not fail if an error is encountered when cloning repos using the examples.py script.

The script should be looked at to determine if it correctly fails when network issues are encountered, otherwise, the build_wrapper job should be looked at.

#### Log: [502 bad gateways.log](https://github.com/ARMmbed/mbed-os/files/2525268/502.bad.gateways.log)

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

adbridge, 2018-10-31:
So it looks like the script tries to clone and if it fails it just prints an error and then continues with the next repo.
```
                for i in range(0, retry):
                    if subprocess.call([repo_info['type'], "clone", repo_info['repo']]) == 0:
                        break
                else:
                    print("ERROR : unable to clone the repo {}".format(name))
```

adbridge, 2018-10-31:
The same is true of these commands also :
1.  do_import(_, config, examples):
2.  do_deploy(_, config, examples):
3. def do_versionning(args, config, examples):

All of these just return 0 no matter what happens inside of the called functions.

closed 2019-02-05 by ciarmcom