ARMmbed/mbed-os #8571: Nrf52 us_ticker_free should also call TASK_SHUTDOWN

NeilMacMullen, 2018-10-29:
### Description
The implementation of us_ticker_free in target_nordic/target_nrf5x/us_ticker.c does not call TASK_SHUTDOWN meaning that the timer does not return to its lowest power mode.  I believe the correct code is...

```
void us_ticker_free(void)
{
    nrf_timer_task_trigger(NRF_TIMER1, NRF_TIMER_TASK_STOP);
    nrf_timer_task_trigger(NRF_TIMER1, NRF_TIMER_TASK_SHUTDOWN); //shutdown for lowest power
    nrf_timer_int_disable(NRF_TIMER1, nrf_timer_compare_int_get(NRF_TIMER_CC_CHANNEL0));
    NVIC_DisableIRQ(TIMER1_IRQn);
    us_ticker_initialized = false;
}
```

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

desmond-blue, 2019-07-30:
Hi @NeilMacMullen , I think you are right, would you be able to send a PR of this proposed fix? Thanks.

NeilMacMullen, 2019-07-30:
@desmond-blue Sorry I don't have time to raise a PR but I think the code I put in the original post addresses the issue.

desmond-blue, 2019-07-30:
@NeilMacMullen, ok, I will raise a PR then, so this issue should be closed.

closed 2019-07-30 by NeilMacMullen