ARMmbed/mbed-os #8597: Need working example code of MCU-to-MCU SPI comm

andrewc-arm, 2018-10-31:
### Description

- What target does this relate to?
    - NXP's K64F and ST's STM32L476
- What toolchain (name + version) are you using?
    - GCC_ARM: 6.3.1 20170620
- What tools (name + version - is it mbed-cli, online compiler or IDE) are you using?
    - mbed-cli: 1.5.0
- What is the SHA of Mbed OS (git log -n1 --oneline)?
    - mbed-os: mbed-os-5.10.2
- Steps to reproduce. (Did you publish code or a test case that exhibits the problem?)
    - For Arm only: there is an internal JIRA case: https://jira.arm.com/browse/IOTCUST-83
    - Use HWs of Mbed-Enabled K64F or STM32L476 as SPI master or SPI slave.
        - Make sure to connect GND, D10~D13 pins between two boards.
    - Use code of 'SPI master' and 'SPI slave' which will be listed below.
    - For STM32L476, it will kernel panic immediately.
    - For K64F, it will only receive '0x00'.
    - We need to make sure STM32L476 will not crash and K64F to send/receive something meaningful.
- SPI master code
```
#include "mbed.h"

SPI spi(D11, D12, D13); // mosi, miso, sclk
DigitalOut cs(D10);
DigitalOut led(LED1);

int main() {
    led = 1;
    // Chip must be deselected
    cs = 1;

    // Setup the spi for 8 bit data, high steady state clock,
    // second edge capture, with a 1MHz clock rate
    spi.format(8,3);
    spi.frequency(1000000);

    int v = 1;
    while(1)
    {   
        led = 0;
        // Select the device by seting chip select low
        cs = 0;
        // Send a byte to receive the data in full duplex mode.
        v = spi.write(v); 
        // Deselect the device
        cs = 1;

        printf("Master v = 0x%X\n", v);
        led = 1;

        wait(5);
    }
}
```

- SPI slave code
```
#include "mbed.h"

SPISlave device(D11, D12, D13, D10); // mosi, miso, sclk, ssel
DigitalOut led(LED1);

int main() {
    led = 1;

    // Setup the spi for 8 bit data, high steady state clock,
    // second edge capture, with a 1MHz clock rate
    device.format(8,3);
    device.frequency(1000000);

    device.reply(0x00);   // Full duplex to read first
    // TODO: change to GPIO handler
    while(1) {
        if (device.receive()) {
            led = 0;
            int v = device.read();   // Read byte from master
            device.reply((v + 1) % 0x100);
            printf("read value: %02X\r\n", v);
            led = 1;
        }
    }
}
```

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

andrewc-arm, 2018-10-31:
Hi. Thanks for the feedback. Unfortunately, I do not have enough bandwidth to follow and test cases. If possible, can mbed-os team kindly test and showcase working example code with Mbed-Enabled boards?

bcostm, 2018-10-31:
Yes :) as well as other peoples in @ARMmbed/team-st-mcd

andrewc-arm, 2018-11-09:
Hi. I just want to leave comment on successful method.
Though os.mbed.com's board's MISO and MOSI pin names are a bit misleading. It's MISO(D11) and MOSI(D12) for SPI master. However, it's MISO(D12) and MOSI(D11) for SPI slave. Therefore, for MCU-to-MCU testing, data pins should be swapped just like UART.