ARMmbed/mbed-os #8626: Update device peripherals for STM32 series

AGlass0fMilk, 2018-11-02:
### Description
Targets in the STM32F4 family have inconsistent "Device Has" peripherals defined. For example, I was working with the STM32F407VG-based discovery board and found that targets.json did not enable "SERIAL_FC" or "SERIAL_ASYNCH". 

I dug a little deeper and found some STM32F4 targets that do enable these peripherals. I also found that all STM32F4 targets share the same serial_api.c HAL implementation, so if one supports these peripherals then they all can. I modified my target's "device_has" field to include "SERIAL_FC" and it worked without issue.

It took me a bit to debug this and I'd like to save everyone time in the future. I think targets need to be gone through (at least in the STM32 family) and peripherals sets should be updated.

### Issue request type

    [ ] Question
    [ ] Enhancement
    [ X ] Bug

bcostm, 2018-11-04:
Hi
The DISCO_F407 is one of the ST boards that has been added by community abd not by ST team in the pasr, so it may have some features/files not well defined compared to other "official" boards. If you already have done the change and tested it ok, why don't you send a PR ? We'll review it. Thanks

AGlass0fMilk, 2019-01-30:
Yeah, I don't primarily work on ST targets so I probably won't get around to this in the near future... It _should_ be documented somewhere so no one else wastes hours of time debugging issues.

I guess anyone using targets that aren't "officially" Mbed-enabled should beware.

jeromecoutant, 2019-01-30:
Please check #9553 
Thx

PR #9553 (2019-02-04): STM32: Enable SERIAL_FC for all
### Description

This fixes #8626 

There is no restriction to enable SERIAL_FC at FAMILY_STM32 level,
and not for all targets 1 by 1.

### Pull request type

    [x] Fix
    [ ] Refactor
    [ ] Target update
    [ ] Functionality change
    [ ] Docs update
    [ ] Test update
    [ ] Breaking change
closed 2019-02-04 by 0xc0170