ARMmbed/mbed-os #8632: RequestsDependencyWarning Message

jflynn129, 2018-11-02:
### Description
When compiling with the latest Mbed-OS (5.10.2), the following warning is generated. What is it and how to get it removed?  I'm building on Windows7, Dos Window.

This occurs every time a compile command is issued.

### Issue request type

    [X ] Question
    [ ] Enhancement
    [ ] Bug

mbed compile -m NUCLEO_L496ZG -t gcc_arm --profile toolchain_debug.json
**[mbed] Auto-installing missing Python modules...**
**c:\python27\lib\site-packages\requests\__init__.py:80: RequestsDependencyWarning: urllib3 (1.23) or chardet (3.0.4) doesn't match a supported version!
  RequestsDependencyWarning)**
[Warning] @,: Compiler version mismatch: Have 4.9.3; expected version >= 6.0.0 and < 7.0.0
Building project azure-iot-mbed-client (NUCLEO_L496ZG, GCC_ARM)

cmonr, 2018-11-06:
HI @jflynn129,

The issue is that the version of gcc-arm-none-eabi that you're using is below the minimum supported version of 6.0.0. To get rid of the warning, all you should need to do is update GCC to v6. 

Note: We currently have capped the supported version to below 7, and are working to catch up to the latest released versions, but this will take some time.

cmonr, 2018-11-06:
It means that while Mbed OS _may_ successfully compile, we can not guarantee that everything will work, because we use GCC 6 to compile and test the OS.

closed 2018-11-07 by jflynn129