ARMmbed/mbed-os #8635: NRF52840: setPhy method is not implemented.

yogeshk19, 2018-11-03:
Description:
setPhy is not implemented in the NRF52840 target and is critical for long range testing, one of the key features in BLE 5. Also the softdevice version S140 is still in alpha,  s140_nrf52840_5.0.0-2.alpha_softdevice.hex in the current mbed-os verison and the current production version of s140 is 6.1.0 and has the implementation of the coded phy. I am hoping the necessary steps can be taken to upgrade the softdevice to production version ASAP.

Target: Nordic nRF52840
Toolchain: mbed cli 1.8.2, arm-gcc
MBED-OS: 5.10.1

How to reproduce:

```
void BleSensor::OnBleConnection(const Gap::ConnectionCallbackParams_t *params)
{
	if(params != NULL) 
	{
		if(params->connectionParams != NULL) 
		{
			FTDigitalLedOut = 1;
	               Gap::PhySet_t txPhys(false, false, true);
			Gap::PhySet_t rxPhys(false, false, true);
			ble_error_t error = m_ble.gap().setPhy(params->handle,&txPhys,&rxPhys,Gap::CodedSymbolPerBit_t::UNDEFINED);

			if(error != BLE_ERROR_NONE) {
				logError("Unable to SetPhy with error code:%d",error);
			}
		}
	}
}
```
Error code returned is 2, which is BLE_ERROR_NOT_IMPLEMENTED

Also looking at the code base for NRF52840,  it is not implemented as mentioned below. 

```
ble_error_t nRF5xGap::setPhy(
    Handle_t connection,
    const ble::phy_set_t* txPhys,
    const ble::phy_set_t* rxPhys,
    CodedSymbolPerBit_t codedSymbol
) {
#ifdef S140
    return BLE_ERROR_NOT_IMPLEMENTED;
#else
    // TODO handle coded symbol once supported by the softdevice.
    ble_gap_phys_t gap_phys = {
        txPhys? txPhys->value() : 0,
        rxPhys? rxPhys->value() : 0
    };

    uint32_t err = sd_ble_gap_phy_update(connection, &gap_phys);

    switch (err) {
        case NRF_SUCCESS:
            return BLE_ERROR_NONE;
        case NRF_ERROR_INVALID_ADDR:
        case BLE_ERROR_INVALID_CONN_HANDLE:
        case NRF_ERROR_INVALID_PARAM:
            return BLE_ERROR_INVALID_PARAM;
        case NRF_ERROR_INVALID_STATE:
            return BLE_ERROR_INVALID_STATE;
        case NRF_ERROR_BUSY:
            return BLE_STACK_BUSY;
        default:
            return BLE_ERROR_UNSPECIFIED;
    }
#endif
}
```

Issue request type
[ ] Question
[x] Enhancement
[x] Bug

cmonr, 2018-11-07:
@yogeshk19 Updated the issue. Please follow the issue template in the future.

yogeshk19, 2018-11-28:
Is there any progress with this issue? This bug is critical for long range testing, one of the key features in BLE 5 and hopefully this fix also incorporates a more current and production version of the Nordic softdevice?

Thanks,
Yogesh

pan-, 2018-11-28:
@yogeshk19 PHY support will be delivered to the NRF52840 with the help of the Cordio link layer (see #8876 ). There is no plan I am aware of regarding the update to a new revision of the Nordic softdevice.

yogeshk19, 2018-11-28:
@pan-  Thanks for the update. I have not used the Cordio feature yet, however I don't understand how would the setPhy method work without the softdevice being updated? Also what happens for scenarios where the MBED BLE API's don't address, that are very specific to the Nordic Softdevice or platform specific? Is there any documentation regarding the Cordio BLE api's that I can get access to?

Thanks,
Yogesh

pan-, 2018-11-29:
@yogeshk19 With the release of Mbed-os 5.11 in few weeks we are adding missing Bluetooth 5 features to our user API. That change include (but is not limited to): 
- Proper connection update management 
- Extended advertising 
- Periodic advertising  

There is no implementation of these APIs based on Softdevice. However for Nordic targets we offer to our users the possibility to use the Cordio Link Layer which in that case completely replace the Softdevice. 

It would be interesting to list use cases that might be not covered by that approach. If you think there is any.

yogeshk19, 2018-11-29:
@pan-  That would be great if we can develop BLE applications irrespective of the soft device or a platform specific API set, Infact that would be a great relief. However I do use the Flash API's for storage and that uses the Soft Device API's I believe. Not sure how that would work when using Cordio Link Layer? 

So I am assuming with the upcoming release in addition to the features one listed above, the setPhy method would be supported for the Nordic Targets?

Thanks,
Yogesh