ARMmbed/mbed-os #8656: SAML21 ticker only fires when set to 1ms. or higher

janjongboom, 2018-11-06:
### Description

I'm trying to update the SAML21 target to use  Mbed OS 5 but I'm running into a weird issue (thanks unit tests!)... When setting a `Timeout`/`Ticker` object to tick with a timeout below 1 ms. the ticker does not fire. I assume it has something to do with the timer setup, but I don't know where to look. If I set the prescaler to 4 in `us_ticker.c` it works (at 500 us), but then everything runs at half speed, so not sure...

### Issue request type

    [X] Question
    [ ] Enhancement
    [ ] Bug

janjongboom, 2018-11-06:
Potentially related: https://github.com/ARMmbed/mbed-os/issues/2598

cmonr, 2019-01-03:
@ARMmbed/mbed-os-core @ARMmbed/mbed-os-hal Thoughts?

mprse, 2019-01-03:
Hi @janjongboom! I checked the PR and I think that you need to adapt us ticker driver to the new standards first. The requirements can be found here:
https://github.com/ARMmbed/mbed-os/blob/51b8d6e59d17748ca477505e77a44a848ceef4c4/hal/us_ticker_api.h#L30-L104

You can run the following tests to check the us ticker driver:
- `tests-mbed_hal-common_tickers`
- `tests-mbed_hal-common_tickers_freq`
- `tests-mbed_hal-us_ticker`

Generally I suggest to start with specifying the counter frequency and size used to handle the us ticker. This is required by the higher layer. If not specified, then default values are used (1MHz/32bits). Example can be found below. You can also use this file as pattern.

https://github.com/ARMmbed/mbed-os/blob/51b8d6e59d17748ca477505e77a44a848ceef4c4/targets/TARGET_NORDIC/TARGET_NRF5x/us_ticker.c#L49-L57

janjongboom, 2019-01-03:
@mprse, Thanks! Will take a look. Hope this will get me a bbit further.

closed 2019-03-15 by janjongboom