ARMmbed/mbed-os #8674: Support for Nordic NRF9160

shankar-vv, 2018-11-08:
### Description

### Issue request type

    [ ] Question
    [ ] Enhancement
    [ ] Bug

shankar-vv, 2018-11-08:
Is there a plan to support the Nordic NRF91 series microcontrollers.
especially NRF9160 which is Cortex-M33.

0xc0170, 2018-11-08:
I updated your first comment (to contain all in description and issue type).

We will review this question

0xc0170, 2018-11-15:
There are currently no plans for supporting NRF91 series microcontrollers.

Contributions are welcome.

AGlass0fMilk, 2019-01-14:
I would also be interested in this. If I get my hands on a nRF9160-DK maybe I can work on porting it sometime 😃

finnss, 2019-10-13:
This would definitely be great!

trowbridgec, 2019-10-14:
I'd be interested too!

0xc0170, 2019-10-14:
I'll close this as it was a question and it was answered. We shall continue discussion about this if needed in forum: https://forums.mbed.com/

closed 2019-10-14 by 0xc0170

finnss, 2019-10-14:
I made a post about support for nRF9160 on the forums [here](https://forums.mbed.com/t/support-for-nordic-nrf9160-dk/5717).