ARMmbed/mbed-os #8678: PSK using TLSSocket

amq, 2018-11-08:
### Description

The [design document](https://github.com/SeppoTakalo/mbed-os/blob/5d1cd2552b7ba3d439f4f087e269fe6a3eb015b4/docs/design-documents/features/connectivity/TLSSocket%20-%20Design%20document.md) of TLSSocket states that there is no PSK mode. Is it planned for the future?

### Issue request type

    [x] Question
    [ ] Enhancement
    [ ] Bug

syhoangmai, 2019-01-07:
Hi, I am also having the same question as I am needing to use PSK mode, but for DTLS instead of TLS.

closed 2019-04-18 by ciarmcom

linlingao, 2019-04-30:
@amq This is an old question. Are you still waiting on an answer?

amq, 2020-03-09:
@linlingao yes, I'm still very interested in the PSK mode for TLSSocket  and DTLSSocket.