ARMmbed/mbed-os #8691: tests-mbed_hal-rtc_time_conv randomly fails on CI

mprse, 2018-11-09:
### Description

Target: `NRF52_DK`
Toolchain : `All`

Test `tests-mbed_hal-rtc_time_conv` randomly fails on CI. This test sends lots of data between host and the device using UART and sometimes the transmission is broken. 

It looks like the issue has been first time caught in PR https://github.com/ARMmbed/mbed-os/pull/8292 and then after the PR has been merged, the test started to fail on CI. 

## Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

0xc0170, 2018-11-09:
> It looks like the issue has been first time caught in PR #8292 and then after the PR has been merged, the test started to fail on CI.

@marcuschangarm Have you experienced any issue ? 

@ARMmbed/mbed-os-hal Can you help reviewing this bug ?

mprse, 2018-11-09:
Yes the RTC is disabled on NRF52_DK. If the RTC is not supported but lp ticker is available, then rtc API is available and is using lp timer. Mainly this is also not the case of this test. This test verifies functions which only supports RTC. These are the time conversion funcions which converts unix timestamp to date/time and vice versa. They do not perform hw operations and the test is executed for targets with and without RTC support. 

The problem of this test is that it sends lots of data to the host. Host verifies if time conversion was valid. I think that changes provided by the mentioned PR have influence on uart transmission since the test was ok before this change has been added. 

Maybe we could create a PR which reverts this change and run morph few times.

marcuschangarm, 2018-11-09:
The PR fixes hardware flow control for the NRF52. But since we don't have hardware flow control on the CI the code is inert during morph test.

NirSonnenschein, 2018-11-14:
Note: this is still happening:
http://mbed-os-logs.s3-website-us-west-1.amazonaws.com/?prefix=logs/7309/3396
Any updates on this issue?

adbridge, 2020-10-13:
As this was last seen in 2018 and we have since migrated the CI to OSS I am going to close this.

closed 2020-10-13 by adbridge