ARMmbed/mbed-os #8695: Non-matching return type in enet_tasklet_disconnect

baregl, 2018-11-09:
### Description

The function declaration of [enet_tasklet_disconnect](https://github.com/ARMmbed/mbed-os/blob/00fcdb87e65a6336318b50822de69dfd84cdd332/features/nanostack/mbed-mesh-api/source/include/enet_tasklet.h#L28) and the [definition](https://github.com/ARMmbed/mbed-os/blob/00fcdb87e65a6336318b50822de69dfd84cdd332/features/nanostack/mbed-mesh-api/source/ethernet_tasklet.c#L259) don't match. The declaration returns `void`, while the definition return `int8_t`. This causes compilation in gcc with lto to fail.

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

elm3, 2018-11-09:
I have created a PR (https://github.com/ARMmbed/mbed-os/pull/8698) to fix this.

baregl, 2018-11-10:
FYI, the arguments also don't match

deece, 2018-12-14:
Partially fixed by 7e7af773ea286618bdce0d55f109508f3e065493, fixed harder by #9100

linlingao, 2019-05-30:
I'm not sure what "fixed harder" means. It seems PRs have been merged. I'll go ahead and close this issue. Please reopen if you don't agree with my assessment.

closed 2019-05-30 by linlingao