ARMmbed/mbed-os #8706: Build failure when building on Windows \ mingw

RonEld, 2018-11-11:
### Description
Build fails, with the following error message:
`[Fatal Error] mbed_fault_handler.c@23,10: platform/mbed_error.h: No such file or directory`
When building on Windows. I have tried both with Windows command and with Git bash ( using Mingw)
The toolchain used is GCC_ARM
I have tried building both on K64F and on NRF52840_DK.

I believe the failure originates from the commit 14a98245111c0ec54781ecf3eda287cdd4a5dc1a 
built using:
~~~
mbed test --compile -m K64F -t GCC_ARM
mbed test --compil -m NRF52840_DK -t GCC_ARM
~~~

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

theotherjimmy, 2018-11-14:
Resloved by #8697, probably

RonEld, 2018-11-14:
@theotherjimmy I have tried to get your fix to check it, but for some reason the commit id is not recognized, and I cannot pull your branch into my local branch for testing.
I am not sure the PR fixes this issue though, according to the fix description, it is about ignored libraries for online compiler.
I was compiling using mbed cli. I *think* it is related to included headers in files, that are part of a path, and the difference between the path separator in Windows and Linux.

theotherjimmy, 2018-11-14:
@RonEld That fix has the side effect/desired effect of correctly handling paths of length 1, specifically `.`. 

>I am not sure the PR fixes this issue though, according to the fix description, it is about ignored libraries for online compiler.

See the discussion in https://github.com/ARMmbed/mbed-os/issues/8716

>I was compiling using mbed cli. I think it is related to included headers in files, that are part of a path, and the difference between the path separator in Windows and Linux.

That's exactly the code that the fix touches, so good guess.

RonEld, 2018-11-14:
@theotherjimmy Thanks for the clarification!

Unfortunately I couldn't test your fix, so I will test it once the fix is merged to confirm.

cmonr, 2018-11-14:
@RonEld Does master not work for you?

This appears to be really similar to https://github.com/ARMmbed/mbed-os/issues/8716, which was resolved with a recent commit.

RonEld, 2018-11-14:
@cmonr I have just updated my master branch, wiht the fix from #8696 and the issue seems to be fixed.
I agree this is a duplicate of #8716, which I have missed when opening this issue

closed 2018-11-14 by RonEld