ARMmbed/mbed-os #8710: Build failure for FUTURE_SEQUANA_M0 with PSOC6_DYNSRM_DISABLE=1

orenc17, 2018-11-12:
### Description

Failure in compilation of FUTURE_SEQUANA_M0 with -DPSOC6_DYNSRM_DISABLE=1 on GCC_ARM
i'm getting undefined reference to DEFAULT_DIVIDER8_RES in targets\TARGET_Cypress\TARGET_PSOC6\psoc6_utils.c line 142

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

@lrusinowicz

MarceloSalazar, 2018-11-20:
discussed with @alzix - this is not a blocker but important to fix at some point soon

MarceloSalazar, 2018-12-04:
@lrusinowicz can you plese look into this?

orenc17, 2018-12-09:
I'm sorry but #9009 does not really fixes the issue as now lines 103 & 105 define the same constant with different values

lrusinowicz, 2018-12-17:
@orenc17 
Yes, I have missed this warning, nevertheless SRM is working for me now. I'm not seeing any undefined references, so it seems like something specific to your project. Could you point me to your code?

orenc17, 2018-12-17:
@lrusinowicz 
The problem now is redefinition.
Try running  `mbed compile -m FUTURE_SEQUANA_M0_PSA`

lrusinowicz, 2018-12-17:
@orenc17
I've just found the redefinition is just a merge issue introduced when my change was merged into the master where your change was already applied to this file. So actually it's different issue, but I will submit a patch, likely tomorrow.
Have you tried removing the first definition already?  Where is the main() for FUTURE_SEQUANA_M0_PSA?

orenc17, 2018-12-18:
@lrusinowicz 
1. when a redefinition occurs the last definition is being used, so your definition works 
2. the main function is in https://github.com/ARMmbed/mbed-os/blob/master/components/TARGET_PSA/spm/COMPONENT_SPE/spm_main.c

closed 2019-01-17 by orenc17

MarceloSalazar, 2019-01-17:
Fixed here: https://github.com/ARMmbed/mbed-os/pull/9155