ARMmbed/mbed-os #8720: LoRaWAN: Uplink Frame Counter Not Incremented After Last Retry of Confirmed Message

mattbrown015, 2018-11-12:
### Description

If the ack for a confirmed message is not received the message will be resent without incrementing the uplink frame counter.  OK :-)

Once the confirmed message retries have been exhausted the message will be discarded and the next new message sent.  AFAICT, the frame counter should be incremented before sending the new message but it isn't.

I set the confirmed message retries to 0 and then ensured the ack's were never received by making the RX window open at the wrong time!  It is then easy to see that every message is sent with the same uplink frame counter.

`LoRaMac::post_process_mcps_req` seems to be at the centre of this but no obvious fix presented itself to me. :-(

CC @hasnainvirk 

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

hasnainvirk, 2018-11-14:
@mattbrown015 This is interesting. Because the premise is that if all the retries are exhausted, that means its highly likely that the base station is not responding and if the node have missed all the acks that means there is something wrong with the node timing. Can you please refer me to a section in the spec where it says that after max number of retries, frame counter should be incremented ?

mattbrown015, 2018-11-14:
Hi @hasnainvirk,

I agree that this is a tricky edge case but believe the point is valid nonetheless.

I don't think LoRaWAN 1.0 covers this scenario explicitly but from "4.3.1.5 Frame counter (FCnt)":

> The end-device shall not reuse the same FCntUp value, except for retransmission, with the same application and network session keys.

Unfortunately "18.1 Uplink Timing Diagram for Confirmed Data Messages" shows the ack being received after the first retry and therefore doesn't illustrate what I'm saying.

AFAICT, the number of retries is configurable from 0 to 254.  I would have thought it is perfectly possible for an ack to be missed if the retries is set to 0 or 1.

As an aside, I have a nitpick! Checking a `uint8_t` against 255 seems a bit pointless.  Is it really the case that 254 retries is acceptable but 255 isn't?  I suspect the type got changed at some point in the distant past.

    lorawan_status_t LoRaWANStack::set_confirmed_msg_retry(uint8_t count)
    {
        if (_device_current_state == DEVICE_STATE_NOT_INITIALIZED) {
            return LORAWAN_STATUS_NOT_INITIALIZED;
        }

        if (count >= MAX_CONFIRMED_MSG_RETRIES) {
            return LORAWAN_STATUS_PARAMETER_INVALID;
        }

        _num_retry = count;

        return LORAWAN_STATUS_OK;
    }

mattbrown015, 2018-11-14:
I've just tried running with confirmed message retries set to 0.  I didn't have to do anything to ensure the downlink where missed, I just left it running and after 6 messages the downlink ack was missed.

The 7th message was transmitted with the same *FCntUp* as the 6th even though retries are set to 0.  TTN thought it was a retry because of the repeated *FCntUp* but the payload was different.

Extract from the TTN Console:

Dir|time|counter|port|  | |
---|-----|--------|----|-|-|
dn|11:53:54| |0| | |
up|11:53:55|5|15| retry confirmed | payload: 01 5B EC 0C A3 80 01 1D |
dn|11:52:54| |0| | |
up|11:52:55|5|15| confirmed | payload: 01 5B EC 0C 67 80 01 1D |

hasnainvirk, 2018-11-15:
In my opinion this should not be an issue. The spec says that it is the node's discretion to either move on i.e., increment the frame counter or don't do it. If all the retries failed, from the node's perspective, it still has not sent the given message and hence the frame counter is not incremented. 
I agree it wouldn't harm to increment the frame counter but I would also argue that not increment it as well brings no harm. NS will still ack it even though it thinks that it is a retry. 

For reference, please check section 4.3.1.3 of the 1.0.2 Spec.

mattbrown015, 2018-11-16:
Unfortunately it does cause us an issue.  We're using FCntUp to differentiate retries and new packets.  The network server and our application server may have received the packet while the end-device hasn't received the ack.  I.e. the network server may receive 2 identical uplinks containing identical data and we want to detect this.  Our application server was designed and tested using a 3rd parity's end-device and that end-device increments the FCntUp in the way I've described.

I'll repeat my citing of LoRaWAN V1.0.2 and add my own emphasis.  From "4.3.1.5 Frame counter (FCnt)": 

> Subsequently FCntUp and 17 FCntDown are **incremented** at the sender side by 1 for each **new** data frame sent in the respective direction.

and:

> The end-device shall **not reuse** the same FCntUp value, except for retransmission, with the 32 same application and network session keys.

hasnainvirk, 2018-11-16:
@mattbrown015 I can buy your argument. It brings no harm in incrementing the counter, so we can do it. I will patch as soon as I can.

mattbrown015, 2018-11-16:
Great! Thanks! :-)

I'm more than willing to help but I took a brief look at the code and thought it might be relatively complicated and 'out of my league'.

Let me know if I can be of any assistance.

mattbrown015, 2018-11-22:
See #8822, specifically [UL FCnt increment in case of retry exhaustion](https://github.com/ARMmbed/mbed-os/pull/8822/commits/3a293b2dde75e4fca59e2e13dff8e3bf47835901).

Just waiting for the PR to get merged. :-)

closed 2018-11-30 by ciarmcom