ARMmbed/mbed-os #8721: Mac address uniquely set prior to running tests?

cmonr, 2018-11-12:
### Description

Do the network interface tests set mac addresses such that IP leases don't collide across multiple devices? Or do the network interfaces not need to be unique?

While looking at https://github.com/ARMmbed/mbed-os/issues/2737, I was able to find MAC addresses that looked suspiciously like VID/PID pairs.

### Issue request type

    [x] Question
    [ ] Enhancement
    [ ] Bug

SeppoTakalo, 2018-11-13:
MAC addresses cannot be set by the applications.

Mbed OS provides macros to supply build time addresses, but those are all zero: https://github.com/ARMmbed/mbed-os/blob/master/platform/mbed_interface.h#L36..L43

So drivers rely on [mbed_mac_address()](https://github.com/ARMmbed/mbed-os/blob/master/platform/mbed_interface.c#L94..L124) which is weak symbol.

I don't know whether platforms supply that or not.

jeromecoutant, 2018-11-13:
In ST side:
https://github.com/ARMmbed/mbed-os/blob/0404701b5f5b9d2aff76432991c74a36bd80742c/features/netsocket/emac-drivers/TARGET_STM_EMAC/stm32xx_emac.cpp#L436

We implement a default function to get a unique MAC address based on the UEID,
but we give customer the possibility to implement their own function in mbed_otp_mac_address()

Regards,

closed 2019-01-18 by TacoGrandeTX