ARMmbed/mbed-os #8724: Build failure while running greentea test

KG-91, 2018-11-13:
### Description

Getting Build failure in my PC, while running green tea test. Let me know how to fix it.
Target: STM32F446RE
Toolchain: ARM

Software and versions
- Windows		    : 8.1 pro – 64 bit
- Python		    : 2.7.10
- Mbed-cli		    : 1.8.2
- Greentea		    : 1.4.0
- Mbedhtrun		    : 1.4.2
- Pip			    : 18.1
- Git Bash(git)		: 2.5.1.windows.1

Attached
1. build failure log
2. log of $pip list

### Issue request type

[build_failure_F446RE.txt](https://github.com/ARMmbed/mbed-os/files/2575142/build_failure_F446RE.txt)
[pip_list.txt](https://github.com/ARMmbed/mbed-os/files/2575143/pip_list.txt)

    [ ] Question
    [ ] Enhancement
    [x] Bug

bcostm, 2018-11-13:
Your build log indicates an old compiler version not supported:

Compiler version mismatch: Have 5.05; expected version >= 5.06 and < 5.07

KG-91, 2018-11-13:
@bcostm 

thanks for your reply
Getting same issue with GCC 6 tool chain also.

[build_failure_GCC_ARM_F446RE.txt](https://github.com/ARMmbed/mbed-os/files/2575374/build_failure_GCC_ARM_F446RE.txt)

KG-91, 2018-11-13:
Working fine with mbed-os-5.9.7 but creating an above issue when upgraded to mbed-os-5.10.3.

0xc0170, 2018-11-13:
Can you run just one test and with verbose and share it ? I can't spot the error in the logs you shared

bcostm, 2018-11-14:
> wondering how directory change salved issue.

Maybe the directory path was too long before ?

KG-91, 2018-11-14:
> > wondering how directory change salved issue.
> 
> Maybe the directory path was too long before ?

Yes, directory path was long. Thanks @bcostm and @0xc0170

closed 2018-11-14 by KG-91