ARMmbed/mbed-os #8726: K82F/KW24D: timeouts in CI in the last 4 runs - various tests

0xc0170, 2018-11-13:
### Description

Tests ( 3381-3383) show various failures for K82F and KW24D. Those I reviewed, show timeout (the test run for more than timeout was set). Most of tests have timeouts 60 seconds.

I run locally mbed hal sleep and mbed hal rtc test on K82F for ARM, all OK.

Links for failed tests:
- https://github.com/ARMmbed/mbed-os/pull/8717#issuecomment-438120808
- https://github.com/ARMmbed/mbed-os/pull/8692#issuecomment-438100171
- https://github.com/ARMmbed/mbed-os/pull/8668#issuecomment-438081966

@ARMmbed/mbed-os-test @OPpuolitaival 

I tested locally

```
mbed test --compile -m K82F -t ARM -n tests-mbed_hal-lp_ticker -DMBED_HEAP_STATS_ENABLED=1 -DMBED_STACK_STATS_ENABLED=1 -DMBED_TRAP_ERRORS_ENABLED=1 -DMBED_ALL_STATS_ENABLED

mbedgt -n tests-mbed_hal-lp_ticker -v -V

+----------+---------------+--------------------------+-----------------------+--------+--------+--------+--------------------+
| target   | platform_name | test suite               | test case             | passed | failed | result | elapsed_time (sec) |
+----------+---------------+--------------------------+-----------------------+--------+--------+--------+--------------------+
| K82F-ARM | K82F          | tests-mbed_hal-lp_ticker | lp ticker glitch test | 1      | 0      | OK     | 0.08               |
| K82F-ARM | K82F          | tests-mbed_hal-lp_ticker | lp ticker info test   | 1      | 0      | OK     | 0.05               |
| K82F-ARM | K82F          | tests-mbed_hal-lp_ticker | lp ticker sleep test  | 1      | 0      | OK     | 0.08               |
+----------+---------------+--------------------------+-----------------------+--------+--------+--------+--------------------+
mbedgt: test case results: 3 OK
mbedgt: completed in 44.09 sec
```

My mbed master SHA: `0404701b5 - (HEAD -> master, upstream/master) Merge pull request #8694 from cmonr/rollup (4 days ago)`

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

0xc0170, 2018-11-13:
I took the binary from 3383 job for mbed_hal-lp_ticker, IAR (one that failed).

```
+----------+---------------+--------------------------+-----------------------+--------+--------+--------+--------------------+
| target   | platform_name | test suite               | test case             | passed | failed | result | elapsed_time (sec) |
+----------+---------------+--------------------------+-----------------------+--------+--------+--------+--------------------+
| K82F-IAR | K82F          | tests-mbed_hal-lp_ticker | lp ticker glitch test | 1      | 0      | OK     | 0.08               |
| K82F-IAR | K82F          | tests-mbed_hal-lp_ticker | lp ticker info test   | 1      | 0      | OK     | 0.05               |
| K82F-IAR | K82F          | tests-mbed_hal-lp_ticker | lp ticker sleep test  | 1      | 0      | OK     | 0.08               |
+----------+---------------+--------------------------+-----------------------+--------+--------+--------+--------------------+
mbedgt: test case results: 3 OK
mbedgt: completed in 44.21 sec
```

@studavekar Please restart these 2 targets (possibly test locally that they work)

0xc0170, 2018-11-13:
As @studavekar mentioned, we need to retest failures in tests (the 2 I shared were unfortunately timeouts, not asserts in tests).

studavekar, 2018-11-13:
@0xc0170 I have cold booted the devices.

jamesbeyond, 2018-11-13:
Hi @studavekar @OPpuolitaival, Quick question.
Can we control the target to cold boot from RaaS or CI? 
Shall we cold boot the target everytime before it starts test?

studavekar, 2018-11-13:
> Hi @studavekar @OPpuolitaival, Quick question.
> Can we control the target to cold boot from RaaS or CI?
> Shall we cold boot the target everytime before it starts test?

@jamesbeyond Austin lab doesn't have that capability, However new Oulu CI has so we are waiting for it to come live.

closed 2018-11-15 by 0xc0170