ARMmbed/mbed-os #8741: [QUESTION] CoAP De-Duplication mechanism testing

daniel-cesarini-tridonic-com, 2018-11-14:
### Description

How is the testing of CoAP De-Duplication mechanism being done? 
Namely, how can you / we test whether setting SN_COAP_DUPLICATION_MAX_MSGS_COUNT to a value greater than 1 is actually causing CoAP packets that are duplicated to being dropped?

I am asking because we want to integrate a specific regression test into our testing suite, and would like to understand what approach is used upstream to cope with this testing?
I believe that the main functionality for de-duplication management is around https://github.com/ARMmbed/mbed-os/blob/83dada3bdc736dad54d882905a15b6bb560abf1d/features/frameworks/mbed-coap/source/sn_coap_protocol.c#L718 .

Some approaches that came to our mind would be to manipulate a sender to actually send duplicated messages, or to change the behavior of a CoAP server (like Leshan) sitting between user applications and end-devices, or to use a tool like dummynet (https://github.com/luigirizzo/dummynet) or mininet (http://mininet.org/) to manipulate the traffic between Leshan and end-devices.

So, finally there are 2 questions:
1) what approach has been used until now by the developers of SN_COAP_ ?
2) what approach do you suggest for us to use to do regression testing on our side?

### Issue request type

    [x] Question
    [ ] Enhancement
    [ ] Bug

FYI: @MarceloSalazar 

Tridonic internal: @markus-becker-tridonic-com @andreas-myka-tridonic-com @javier-moreno-tridonic-com

daniel-cesarini-tridonic-com, 2018-11-14:
FYI: @SeppoTakalo @anttiylitokola (as I saw you committed to this directory).

daniel-cesarini-tridonic-com, 2018-11-23:
Ok. Thanks you! ;-)

FYI: @Tanganelli

MarceloSalazar, 2018-12-10:
@markus-becker-tridonic-com can you review and comment on the status of this ticket?

markus-becker-tridonic-com, 2019-01-08:
our testing with coapthon is on-going.

closed 2019-04-18 by ciarmcom

linlingao, 2019-04-30:
@daniel-cesarini-tridonic-com Please reopen this issue if you have further questions.