ARMmbed/mbed-os #8752: lots of sporatic network test failures in CI

NirSonnenschein, 2018-11-15:
### Description
Tracking issue for latest test failures we noticed in recent PR.

the CI is currently suffering from lots of spurious networking issues failing all the recent test runs:
the latest 4 runs failed:
http://mbed-os-logs.s3-website-us-west-1.amazonaws.com/?prefix=logs/8647/3408
http://mbed-os-logs.s3-website-us-west-1.amazonaws.com/?prefix=logs/8190/3409
http://mbed-os-logs.s3-website-us-west-1.amazonaws.com/?prefix=logs/8727/3410
http://mbed-os-logs.s3-website-us-west-1.amazonaws.com/?prefix=logs/8697/3411

sample test failures for one of the jobs:

ARCH_PRO-GCC_ARM.tests-netsocket-udp.UDPSOCKET_ECHOTEST_NONBLOCKARCH_PRO-GCC_ARM.tests-netsocket-tcp.TCPSOCKET_THREAD_PER_SOCKET_SAFETYK64F-IAR.tests-netsocket-udp.UDPSOCKET_ECHOTEST_NONBLOCKK64F-IAR.tests-netsocket-tcp.TCPSOCKET_ECHOTEST_NONBLOCKK64F-IAR.tests-netsocket-tcp.TCPSOCKET_THREAD_PER_SOCKET_SAFETYNUCLEO_F207ZG-GCC_ARM.tests-netsocket-tcp.TCPSOCKET_ECHOTESTNUCLEO_F207ZG-GCC_ARM.tests-netsocket-tcp.TCPSOCKET_ECHOTEST_NONBLOCKNUCLEO_F207ZG-GCC_ARM.tests-netsocket-tcp.TCPSOCKET_THREAD_PER_SOCKET_SAFETYNUCLEO_F207ZG-GCC_ARM.tests-netsocket-udp.UDPSOCKET_ECHOTEST_NONBLOCKNUCLEO_F429ZI-GCC_ARM.tests-netsocket-tcp.TCPSOCKET_ECHOTESTNUCLEO_F429ZI-GCC_ARM.tests-netsocket-udp.UDPSOCKET_ECHOTEST_NONBLOCKNUCLEO_F207ZG-IAR.tests-netsocket-tcp.TCPSOCKET_ECHOTESTNUCLEO_F207ZG-IAR.tests-netsocket-tcp.TCPSOCKET_ECHOTEST_NONBLOCKNUCLEO_F207ZG-IAR.tests-netsocket-tcp.TCPSOCKET_THREAD_PER_SOCKET_SAFETYNUCLEO_F207ZG-IAR.tests-netsocket-udp.UDPSOCKET_ECHOTEST_NONBLOCKNUCLEO_F746ZG-GCC_ARM.tests-netsocket-tcp.TCPSOCKET_ECHOTESTNUCLEO_F746ZG-GCC_ARM.tests-netsocket-tcp.TCPSOCKET_ECHOTEST_NONBLOCKNUCLEO_F746ZG-GCC_ARM.tests-netsocket-tcp.TCPSOCKET_THREAD_PER_SOCKET_SAFETYNUCLEO_F746ZG-GCC_ARM.tests-netsocket-udp.UDPSOCKET_ECHOTEST_NONBLOCKNUCLEO_F429ZI-IAR.tests-netsocket-tcp.TCPSOCKET_ECHOTEST_NONBLOCKNUCLEO_F429ZI-IAR.tests-netsocket-tcp.TCPSOCKET_THREAD_PER_SOCKET_SAFETYNUCLEO_F429ZI-IAR.tests-netsocket-udp.UDPSOCKET_ECHOTEST_NONBLOCKNUCLEO_F429ZI-IAR.tests-netsocket-tcp.TCPSOCKET_ECHOTESTK64F-GCC_ARM.tests-netsocket-tcp.TCPSOCKET_ECHOTESTK64F-GCC_ARM.tests-netsocket-tcp.TCPSOCKET_ECHOTEST_NONBLOCKK64F-GCC_ARM.tests-netsocket-tcp.TCPSOCKET_THREAD_PER_SOCKET_SAFETYK64F-GCC_ARM.tests-netsocket-udp.UDPSOCKET_ECHOTEST_NONBLOCKNUCLEO_F429ZI-ARM.tests-netsocket-tcp.TCPSOCKET_ECHOTESTNUCLEO_F429ZI-ARM.tests-netsocket-tcp.TCPSOCKET_ECHOTEST_NONBLOCKNUCLEO_F429ZI-ARM.tests-netsocket-tcp.TCPSOCKET_THREAD_PER_SOCKET_SAFETYNUCLEO_F429ZI-ARM.tests-netsocket-udp.UDPSOCKET_ECHOTEST_NONBLOCKNUCLEO_F746ZG-IAR.tests-netsocket-tcp.TCPSOCKET_ECHOTEST_NONBLOCKNUCLEO_F746ZG-IAR.tests-netsocket-tcp.TCPSOCKET_THREAD_PER_SOCKET_SAFETYNUCLEO_F746ZG-IAR.tests-netsocket-udp.UDPSOCKET_ECHOTEST_NONBLOCKK64F-ARM.tests-netsocket-tcp.TCPSOCKET_ECHOTEST_NONBLOCKK64F-ARM.tests-netsocket-tcp.TCPSOCKET_THREAD_PER_SOCKET_SAFETYK64F-ARM.tests-netsocket-udp.UDPSOCKET_ECHOTEST_NONBLOCKARCH_PRO-ARM.tests-netsocket-udp.UDPSOCKET_ECHOTEST_NONBLOCKARCH_PRO-ARM.tests-netsocket-tcp.TCPSOCKET_ECHOTESTNUCLEO_F207ZG-ARM.tests-netsocket-tcp.TCPSOCKET_ECHOTESTNUCLEO_F207ZG-ARM.tests-netsocket-tcp.TCPSOCKET_ECHOTEST_NONBLOCKNUCLEO_F207ZG-ARM.tests-netsocket-tcp.TCPSOCKET_THREAD_PER_SOCKET_SAFETYNUCLEO_F207ZG-ARM.tests-netsocket-udp.UDPSOCKET_ECHOTEST_NONBLOCKARCH_PRO-IAR.tests-netsocket-udp.UDPSOCKET_ECHOTEST_NONBLOCKNUCLEO_F746ZG-ARM.tests-netsocket-tcp.TCPSOCKET_ECHOTEST_NONBLOCKNUCLEO_F746ZG-ARM.tests-netsocket-tcp.TCPSOCKET_THREAD_PER_SOCKET_SAFETYNUCLEO_F746ZG-ARM.tests-netsocket-udp.UDPSOCKET_ECHOTEST_NONBLOCK
--

This seems to be sporadic as it doesn't fail for all boards all the time but failure rate is very high.
the PRs tested are unrelated to the failure. Seems to be a test setup issue.

### Issue request type

    [ ] Question
    [ ] Enhancement
    [ X ]

NirSonnenschein, 2018-11-15:
After a few hours of interruption this stopped happening for some unknown reason

cmonr, 2018-11-20:
For reference, it appears that this will be resolved once https://github.com/ARMmbed/mbed-os/issues/8715 is as well.