ARMmbed/mbed-os #8774: Cellular: support for setting baud rate and enabling flow control on the target device

mfatiga, 2018-11-16:
### Description

Support for setting baud rate and enabling flow control on the target modem device is required for more stable UART communication with the device.
Flow control should be enabled automatically when MDMCTS and MDMRTS are set, as is enabled for the UART instance communicating with that device.

Loosely related to #8714.

@kjbracey-arm

### Issue request type

    [ ] Question
    [x] Enhancement
    [ ] Bug

AriParkkila, 2018-11-30:
@mfatiga this is now fixed in Mbed OS master and it's integrated in the release 5.11. Can we close this issue?

closed 2018-11-30 by 0xc0170

AriParkkila, 2018-11-30:
@mfatiga please see `CellularDevice::get_default_instance()` in https://github.com/ARMmbed/mbed-os/blob/master/features/cellular/framework/device/CellularDevice.cpp

mfatiga, 2018-11-30:
@AriParkkila This is not what I meant by this issue. This is only enabling flow control on the microcontroller UART, not on the modem device itself. For example, for the QUECTEL BG96 it would be something like this, implemented in the `init_module` function:
```
#if defined (MDMRTS) && defined (MDMCTS)
atHandler->cmd_start("AT+IFC=2,2");
atHandler->cmd_stop();
atHandler->resp_start();
atHandler->resp_stop();
if (atHandler->get_last_error() == NSAPI_ERROR_OK) {
    tr_info("Cellular device - enabled flow control.");
} else {
    tr_warn("Cellular device - failed to enable flow control.");
}
#endif
```

Also, setting the baud rate on the target device is also not implemented. For the QUECTEL BG96 device, this would be done by using `AT+IPR=57600`, and I see no mention of that AT command anywhere in the code. This has to somehow be integrated in the `init_module` function, so the baud rate is set as soon as possible to achieve stable UART communication. The default baud rate (the one from the factory) and the wanted baud rate (that is set upon module initialization) should be configurable.

AriParkkila, 2018-11-30:
@mfatiga unfortunately that's not supported. You need to setup it in production or in your application (e.g. via cellular AT handler coming in #8749).

mfatiga, 2018-11-30:
@AriParkkila Yes, that would be good enough for configuring the baud rate, but then the UART instance should also be accessible externally. 
As for the flow control, I think that automatically enabling flow control on the actual modem device when you have defined MDMCTS and MDMRTS would be much better.

AriParkkila, 2018-11-30:
@mfatiga `ATHandler::get_file_handle` is actually UART so you could use that (in the near future) to `UARTSerial::set_flow_control`.