ARMmbed/mbed-os #8781: RTC Problem on STM32F407VG

masoudr, 2018-11-16:
### Description
I'm using an STM32F407VG board. I can set RTC time with mbed time library and I can read the time when the system has power. When I remove the power, It seems that the time doesn't proceed and I always get the last value for the time.
I've tested this function on Seeed Arch Max board (based on STM32F407VE) and it is fine. Do I need to add some parameters in `targets.json` file?

### Issue request type
    [ ] Question
    [ ] Enhancement
    [* ] Bug

jeromecoutant, 2018-11-19:
Hi
RTC is based on an external 32k clock called LSE.
Is this LSE well connected and powered on your board ?

masoudr, 2018-11-19:
@jeromecoutant. Somehow I fixed the problem by setting `"lse_available": 1` in `targets.json` file. But I think it should be well documented at least in Mbed OS Time API.

jeromecoutant, 2018-11-19:
> But I think it should be well documented at least in Mbed OS Time API.

This is not part of MBED API, this is only defined in STM32 level:

    "FAMILY_STM32": {
        "config": {
            "lse_available": {
                "help": "Define if a Low Speed External xtal (LSE) is available on the board (0 = No, 1 = Yes). If Yes, the LSE will be used to clock the RTC, LPUART, ... otherwise the Low Speed Internal clock (LSI) will be used",
                "value": "1"
            },

You can then see that default value is already 1

masoudr, 2018-11-19:
@jeromecoutant You are right but it is overridden and set to zero for almost every stm32 based boards on `targets.json` file.

masoudr, 2019-06-21:
I still have some problem with RTC. Currently, I've connected an external 32k external oscillator to my ST Board and set `lse_available: 1` in my config file (`MBED_CONF_TARGET_LSE_AVAILABLE` set to true). But the clock gets behind e.g when I set current time after 3 4 hours it gets about 20 seconds behind. I also tried to call `rtc_init` but I get `not defined function` on compiler output. I checked the `rtc_api.c` file and found the function there. But why I can't use it?
I'm thinking that it is still using the internal clock for RTC (LSI). How can I check that what clock source for RTC is active?

jeromecoutant, 2019-06-24:
> How can I check that what clock source for RTC is active?

Something like that?

````
#include "stm32f4xx_ll_rcc.h"

    switch (LL_RCC_GetRTCClockSource()) {
        case LL_RCC_RTC_CLKSOURCE_NONE:
// RTC is not configured
            break;
        case LL_RCC_RTC_CLKSOURCE_LSE:
// LSE is configured
            break;
        case LL_RCC_RTC_CLKSOURCE_LSI:
// LSI is configured
            break;
        default:
            break;
    }
````

masoudr, 2019-06-24:
@jeromecoutant Thanks, I get `LSE is configured` So it seems that It works fine. But I was a little disappointed about the precision of RTC, I think it is not a reliable source to handle date time.

closed 2019-06-24 by masoudr