ARMmbed/mbed-os #8783: Using DEVICE_LPTICKER on STM32L486

TacoGrandeTX, 2018-11-16:
### Description

@jeromecoutant Could we please get some clarification on the removal of LPTICKER for some STM32L4 targets in https://github.com/ARMmbed/mbed-os/commit/c31a7c22774af7d315dc011b8f820f9b5361be7a ?

Are there precautions that can be taken to still make use of DEVICE_LPTICKER?  

We have a customer interested in using MBED_TICKLESS with DEVICE_LPTICKER on a Advantech WISE-1570 (STM32L486-based) and they are experiencing a hang when normal sleep is entered with ThisThread::sleep_for() for 8 seconds or more.   

This is only observed on Mbed OS v5.10.x.  It's okay on Mbed OS v5.9 with Thread::wait().  

### Issue request type

    [X] Question
    [ ] Enhancement
    [ ] Bug

TacoGrandeTX, 2018-11-29:
@jeromecoutant Additional testing has shown that increasing LPTICKER_DELAY_TICKS to 4 prevents the hang.

jeromecoutant, 2018-12-04:
Thx @TacoGrandeTX for this report

Maybe I should set this 4 tick value for all targets.
Will start tests in TICKLESS

@c1728p9 @LMESTM

jeromecoutant, 2018-12-18:
New LPTICKER_DELAY_TICKS value is part of #9134

screamerbg, 2019-01-10:
@TacoGrandeTX Does this answer your question? Mbed OS 5.11.1 contains patch to address this.

TacoGrandeTX, 2019-01-10:
@screamerbg Yes - will close.  Thanks for the followup.

closed 2019-01-10 by TacoGrandeTX