ARMmbed/mbed-os #8786: I2C write issue with AT24C EEPROM TIMEOUT or error in i2c_write HAL_I2C_ErrorCallback:4, index=0 on STM32L4

Hoel, 2018-11-17:
Hi, i encounter a strange issue with MBED and a AT24C64 I2C EEPROM. The issue only appear on I2C write operation and only for the EEPROM, not on other I2C peripherals. The target is STM32l4, the I2C port used is PA9, PA10 with external 4.7K pullups.

I tried two different AT24C64 EEPROM with same result, for good measure i also tested an AT24C04, same error, so EEPROM error is not the cause here. There is a bunch of other I2C peripheral on this bus, none of these have any issue / error during write or read operations. To ensure it was not a incompatibiity issue i tested with an unpopulated PCB, so the only peripheral on the bus was this EEPROM, same issue. The I2C address is correct (incorrect address leads to NACK).

When the EEPROM is written there is a debug message from MBED API : TIMEOUT or error in i2c_write and the HAL error callback is called with a cause 4 and index 0. Depite the error, EEPROM is actually written correctly (checked by reading the content back, all data written on EEPROM are correct). There is no error on read operations.

I captured 4 write + 4 read operations with logic analyzer, (data 0x20,0x21,0x22,0x23 written at address 0x0000,0x0001,0x0002,0x0003 and read back). The weird look of the trace is due to the 2000uS delay added in I2C_API I2C_write function to avoid HAL error.

[image]

￼

first write operation

[image]
[image]

repeated 4 times in total.

first read operation

[image]
[image]

reapeated 4 times in total

Here is the 2000uS delay i added in I2C_API::i2c_write to avoid the HAL error: 

[image]

With this modification there is no error, however this is not an option nor a fix since it does affects the whole I2C API effectively adding a 2ms delay after each I2C write so all peripherals which have no problem otherwise are slowed down which is not possible.

I have tested with I2C bus frequency from 100KHz to 1Mhz, and I2C mode ASYNC set to 1 or 0 (so with DMA or IRQ) no difference, the error is always thrown.

Here is the debug trace on write operation:

[EEPROM] ID:AT24C64 address:0xa0 size:8192 bytes
HAL_I2C_ErrorCallback:4, index=0
    TIMEOUT or error in i2c_write
    timeout:4/6000uS
    xfer:FIRST_AND_LAST_FRAME
    event:I2C_EVENT_ERROR
HAL_I2C_ErrorCallback:4, index=0
    TIMEOUT or error in i2c_write
    timeout:4/6000uS
    xfer:FIRST_AND_LAST_FRAME
    event:I2C_EVENT_ERROR
[EEPROM] ok - W/R pass
So the timout counter is only 4 uS on 6000 -> no timeout issue.

the xfer flag is FIRST_AND_LAST_FRAME -> what it should be

I2C event flag is EVENT_ERROR -> wrong, it should be I2C_EVENT_TRANSFER_COMPLETE.

When i add the 2000uS delay, then the I2C event flag is I2C_EVENT_TRANSFER_COMPLETE.

Since the data are correctly written despite the error wite/read test result is OK.

Given the cross checks already made, i am not sure what to try or investigate now, this driver had been written on the Nucleo L476 board and there was no issue as far as i can recall, that said i am not sure the I2C debug was enabled, also the I2C bus was not the same (not PA9-PA10).

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

0xc0170, 2018-11-19:
@ARMmbed/team-st-mcd Please rview

mattbrown015, 2018-11-19:
It may be completely unrelated but...

I see something similar when writing to our NFC tag (M24SR).  In my case it happens when performing a 0 byte write which is used to get the ack from the last transaction.  It is repeatable but I have to do several hundred write to get it to happen.

I assumed the problem was with the NFC tag and still think it probably is but I'll be interested to see if anything comes of this issue.

My target is a STM32L442KC.

LMESTM, 2018-11-19:
hi @Hoel,
Could you also get a complete trace analyzer snapshot in case of a failure case ? 
Also referring to @mattbrown015 comment, do you see request for 0 byte read or write ?

mattbrown015, 2018-11-19:
> Also referring to @mattbrown015 comment, do you see request for 0 byte read or write ?

Oh, on a write.  I didn't write the code but I believe it is polling to see if the tag is ready.  Something like, write 0 bytes and if no I2C ack then the NFC tag is still busy.  If an I2C ack is received the NFC tag has finished the command.  Occasionally we get stuck in the polling loop and no I2C ack is every received.

LMESTM, 2018-11-19:
@mattbrown015 
if no ACK is ever received from your device, maybe the device has crashed ? 
Note that the SYNC API supports the 0 case and this has been tested in the past. In the case of ASYNCH API implementation I'm not sure ...  

In the past there was a precise check for this and it was removed by a user in commit 251459e8aab8966ce45179cc975ccaebc761fcde. the code was explicitely calling HAL_I2C_IsDeviceReady

```
+    int count = I2C_ERROR_BUS_BUSY, ret = 0;
     uint32_t timeout = 0;
 
+    if((length == 0) || (data == 0)) {
+        if(HAL_I2C_IsDeviceReady(handle, address, 1, 10) == HAL_OK)
+            return 0;
+        else
+            return I2C_ERROR_BUS_BUSY;
+    }
```

In case we think that the problems you are reporting are related to the readyness of the device, it might be worth suing this explicit test again and see if that helps.

@Hoel  can you share a link to the AT24C64 driver code that you're using ?

mattbrown015, 2018-11-19:
> if no ACK is ever received from your device, maybe the device has crashed ?

Yes, agreed, and at present I think that is the most likely scenario.

Sorry, I didn't mean to distract this issue from @Hoel's problem.  Mainly I just wanted to watch what happens on the issue while making a passing comment that I saw something similar.

Hoel, 2018-11-19:
Helo,
Thanks for the replies.
I had no chance to work in the issue this morning, i will take some time in the afternoon.

However more test have been done after i posted the ticket. Also i rewrote the write, poll and read function with single packet I2C read/write commands wich gave more informations on what is going on.

The problem comes from the polling required to know when the byte writing is finished on the AT24C64. This mecanism apparently doesnt work as it should, not sure why yet.

when i poll using write address as advised by the datasheet (I2C address with R/W bit set to 0, 0xA0 here) it return NAKs for few times, then, at about half the writing time (1.5mS in total for 1 byte write in my case) everything hangs, even before the STOP is issued, and MCU is halted forever (?).

I tried to do the polling using the read address (I2C address with R/W bit set to 1, 0xA1 in this case), it kinda works (no halt) however the problem of using the read address is that we have to issue a read command once ACK is received otherwise the I2C bus is held. 

I will send various logic traces very soon to illustrate these two cases.
At first i want to get it working with single byte I2C commands, i suspect the multi-byte write/read commands will not work with thie eeprom, due to timeouts and weird behavior of the EEPROM/MCU when polling after write.

Also, if instead of polling i just set a 1.5 mS delay after the write command everything works fine, i can write the byte and read it afterwards without hang or timeouts, of course this option is not very practical and i wont use it unless there is no way to get the polling mecanism working.

this works, but it is not a solution
```
	cmd[0]=0x00;
	cmd[1]=0x00;
	cmd[2]=0x88;
	i2c.write((int)0xa0,(char *)cmd,3,0);

	wait(0.0015); 			       // 1.5mS is required to write 1 Byte

	i2c.start();
	i2c.write( 0xa0 ); 	 		// device addr + write bit
	i2c.write( 0x00 ); 	  		// word addr lsb
	i2c.write( 0x00 );  		        // word addr msb
	i2c.stop();  	   			// stop

	i2c.start();       			// start
	i2c.write( 0xa1);  			// device addr + read bit
	uint8_t value = i2c.read(1);    // read
	i2c.stop();				 // stop
```

this doesnt work, hangs at middle of the write
```
	cmd[0]=0x00;
	cmd[1]=0x00;
	cmd[2]=0x88;
	i2c.write((int)0xa0,(char *)cmd,3,0);

	while(ack != 0) { i2c.start(); ack = i2c.write( 0xa0 );  i2c.stop(); }

	i2c.start();
	i2c.write( 0xa0 ); 	 		// device addr + write bit
	i2c.write( 0x00 ); 	  		// word addr lsb
	i2c.write( 0x00 );  		        // word addr msb
	i2c.stop();  	   			// stop

	i2c.start();       			// start
	i2c.write( 0xa1);  			// device addr + read bit
	uint8_t value = i2c.read(1);    // read
	i2c.stop();				 // stop
```

LMESTM, 2018-11-19:
Hi,

I would rather suggest to use the multi-byte API. The I2C driver and EEPROM drivers are regularly tested on STM targets using ci-test-shield program. https://github.com/ARMmbed/ci-test-shield

The reference eeprom driver that it uses is here: 
https://os.mbed.com/users/mbed_official/code/I2CEeprom/file/973c4289c44c/I2CEeprom.cpp/

The polling logic in here is
```
void I2CEeprom::waitForWrite() {
    // The chip doesn't ACK while writing to the actual EEPROM
    // so loop trying to do a zero byte write until it is ACKed
    // by the chip.
    while (m_i2c.write(m_i2cAddress, 0, 0) != 0) {
        // Wait for ack.
        wait_ms(1);
    }
}
```
I would suggest to try to re-use this piece of code, using multi-byte API and the return value it provides. you may reduce the wait loop to match your own needs, as it seems 1,5ms is the time it takes the eeprom here to be ready for next write.

Hoel, 2018-11-19:
HI,
@LMESTM OK i will try the suggested waitForWrite and let you know the result.
i used two different drivers for the EEPROM with the same result (timeouts)
They are attached.
[EEPROM.zip](https://github.com/ARMmbed/mbed-os/files/2595815/EEPROM.zip)

mattbrown015, 2018-11-19:
_Interestingly_ that's very similar to the M24SR driver that I'm using, [M24SR](https://os.mbed.com/teams/ST/code/M24SR/).

    M24SR::StatusTypeDef M24SR::M24SR_IO_PollI2C(void) {
        int status = 1;
        while (status != 0 ) {
            //send the device address and wait to recevie an ack bit
            status = dev_I2C.write(address,NULL,0);
        }
        return M24SR_SUCCESS;
    }

I think I tried adding a delay.  We shall see how well it works for @Hoel.

I was getting stuck in the infinite loop when the device didn't ack.

I added a counter and quit if an arbitrary number of retries is exceeded.

Does anyone have a better approach?

It does feel like @Hoel  and I are following the same road.

LMESTM, 2018-11-19:
> I think I tried adding a delay. We shall see how well it works for @Hoel.

@mattbrown015 the device may need a delay between various repeating requests  ?

Hoel, 2018-11-19:
@LMESTM
as told intially, despite the timeout error messages it works, if i disable the debug it most likely will still work (i will check and confirm). 

So do you mean it is the expected behaviour to have timeout error messages when using the I2C driver with a peripheral that relies on this wait-for-write mecanism?
That was not obvious to me when i see the timeout is not ellapsed and this is is an "error in I2C write"

LMESTM, 2018-11-19:
@Hoel  :-) 
I first thought that you activated DEBUG prints in I2C driver because you were facing issues ...

The **TIMEOUT or error in i2c_write** is a geneic message that is useful when debugging. 
the **HAL_I2C_ErrorCallback:4, index=0** is more precise for checking actual error. 
The error ID 4 is the one from STM32 HAL defined here: 
https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_STM/TARGET_STM32L4/device/stm32l4xx_hal_i2c.h#L184
which means acknowledge failure ... 
... which is indeed expected here when waiting for the device to be ready. 

The I2C driver can't tell whether the error is expected or not, so it will display the debug warning.

On the other hand the eeprom driver will expect this, and will not consider it an error, which is why this work okay as I see it now ...

Hoel, 2018-11-19:
@LMESTM
OK it makes sense. In fact debug was enabled while writing driver for another perpheral, when later on i tested the EEPROM debug was still enabled and so i got these error message.
Right, this ticket will be closed now.

closed 2018-11-19 by Hoel