ARMmbed/mbed-os #8792: Configurable SPIF debug message level ?

yplam, 2018-11-19:
### Description

The components in mbed-os/components/storage/blockdevice/ folder use the global tr_info(), tr_debug() directly to print debug messages. It will be so many SPIF info messages if use the filesystem a lot. What about adding a configure option like littlefs mbed-os/features/storage/filesystem/littlefs/littlefs/lfs_util.h:

```
// Logging functions
#if !defined(LFS_NO_INFO) && MBED_LFS_ENABLE_INFO
#define LFS_INFO(fmt, ...)  printf("lfs info:%d: " fmt "\n", __LINE__, __VA_ARGS__)
#elif !defined(LFS_NO_INFO) && !defined(MBED_LFS_ENABLE_INFO)
#define LFS_INFO(fmt, ...)  debug("lfs info:%d: " fmt "\n", __LINE__, __VA_ARGS__)
#else
#define LFS_INFO(fmt, ...)
#endif

#if !defined(LFS_NO_DEBUG) && MBED_LFS_ENABLE_DEBUG
#define LFS_DEBUG(fmt, ...) printf("lfs debug:%d: " fmt "\n", __LINE__, __VA_ARGS__)
#elif !defined(LFS_NO_DEBUG) && !defined(MBED_LFS_ENABLE_DEBUG)
#define LFS_DEBUG(fmt, ...) debug("lfs debug:%d: " fmt "\n", __LINE__, __VA_ARGS__)
#else
#define LFS_DEBUG(fmt, ...)
#endif

#if !defined(LFS_NO_WARN) && MBED_LFS_ENABLE_WARN
#define LFS_WARN(fmt, ...)  printf("lfs warn:%d: " fmt "\n", __LINE__, __VA_ARGS__)
#elif !defined(LFS_NO_WARN) && !defined(MBED_LFS_ENABLE_WARN)
#define LFS_WARN(fmt, ...)  debug("lfs warn:%d: " fmt "\n", __LINE__, __VA_ARGS__)
#else
#define LFS_WARN(fmt, ...)
#endif

#if !defined(LFS_NO_ERROR) && MBED_LFS_ENABLE_ERROR
#define LFS_ERROR(fmt, ...) printf("lfs error:%d: " fmt "\n", __LINE__, __VA_ARGS__)
#elif !defined(LFS_NO_ERROR) && !defined(MBED_LFS_ENABLE_ERROR)
#define LFS_ERROR(fmt, ...) debug("lfs error:%d: " fmt "\n", __LINE__, __VA_ARGS__)
#else
#define LFS_ERROR(fmt, ...)
#endif

// Runtime assertions
#if !defined(LFS_NO_ASSERT) && MBED_LFS_ENABLE_ASSERT
#define LFS_ASSERT(test) assert(test)
#elif !defined(LFS_NO_ASSERT) && !defined(MBED_LFS_ENABLE_ASSERT)
#define LFS_ASSERT(test) MBED_ASSERT(test)
#else
#define LFS_ASSERT(test)
#endif
```

### Issue request type

    [ ] Question
    [ x ] Enhancement
    [ ] Bug

0xc0170, 2018-12-06:
@jupe Please review

yplam, 2018-12-07:
Hi @yossi2le 
Thanks for your reply. Good to see the upcoming feature.

closed 2018-12-07 by yplam