ARMmbed/mbed-os #8805: tests-mbed_drivers-lp_timer.Tes tests fails on K64F

studavekar, 2018-11-19:
### Description

This failure is seen multiple times in CI system, created an issue to track it. May be worn out flash? 

**Target**
K64F

**Toolchain:**
GCC_ARM| ARM|IAR

**mbed-os sha:**

9aef9d3 Merge pull request #8671 from mray19027/uart_break

**Steps to reproduce**

Use command run :  mbed test -m K64F -t GCC_ARM  -n tests-mbed_drivers-lp_time  -v

Failure 

```
[1542618111.29][GLRM][TXD] mbedmbedmbedmbedmbedmbedmbedmbedmbedmbed
[1542618111.29][CONN][INF] sending up to 10 __sync packets (specified with --sync=10)
[1542618111.29][CONN][INF] sending preamble 'f1ff69b6-f687-466f-8d71-98f3725eeecf'
[1542618111.30][GLRM][TXD] {{__sync;f1ff69b6-f687-466f-8d71-98f3725eeecf}}
[1542618112.46][CONN][RXD] mbedmbedmbedmbedmbedmbedmbedmbed
[1542618112.46][CONN][INF] found SYNC in stream: {{__sync;f1ff69b6-f687-466f-8d71-98f3725eeecf}} it is #0 sent, queued...
[1542618112.46][HTST][INF] sync KV found, uuid=f1ff69b6-f687-466f-8d71-98f3725eeecf, timestamp=1542618112.456993
[1542618112.56][CONN][RXD] >>> Running 9 test cases...
[1542618112.56][CONN][INF] found KV pair in stream: {{__version;1.3.0}}, queued...
[1542618112.56][CONN][INF] found KV pair in stream: {{__timeout;15}}, queued...
[1542618112.56][HTST][INF] DUT greentea-client version: 1.3.0
[1542618112.56][CONN][INF] found KV pair in stream: {{__host_test_name;default_auto}}, queued...
[1542618112.56][HTST][INF] setting timeout to: 15 sec
[1542618112.56][HTST][INF] host test class: '<class 'mbed_host_tests.host_tests.default_auto.DefaultAuto'>'
[1542618112.56][HTST][INF] host test setup() call...
[1542618112.56][HTST][INF] CALLBACKs updated
[1542618112.56][HTST][INF] host test detected: default_auto
[1542618112.66][CONN][INF] found KV pair in stream: {{__testcase_name;Test: LowPowerTimer - stopped after creation.}}, queued...
[1542618112.95][CONN][INF] found KV pair in stream: {{__testcase_name;Test: LowPowerTimer - measure time accumulation.}}, queued...
[1542618112.95][CONN][INF] found KV pair in stream: {{__testcase_name;Test: LowPowerTimer - reset.}}, queued...
[1542618112.95][CONN][INF] found KV pair in stream: {{__testcase_name;Test: LowPowerTimer - start started timer.}}, queued...
[1542618112.97][CONN][INF] found KV pair in stream: {{__testcase_name;Test: LowPowerTimer - float operator.}}, queued...
[1542618112.97][CONN][INF] found KV pair in stream: {{__testcase_name;Test: LowPowerTimer - time measurement 1 ms.}}, queued...
[1542618113.06][CONN][INF] found KV pair in stream: {{__testcase_name;Test: LowPowerTimer - time measurement 10 ms.}}, queued...
[1542618113.16][CONN][RXD] 
[1542618113.16][CONN][INF] found KV pair in stream: {{__testcase_name;Test: LowPowerTimer - time measurement 100 ms.}}, queued...
[1542618113.16][CONN][INF] found KV pair in stream: {{__testcase_name;Test: LowPowerTimer - time measurement 1 s.}}, queued...
[1542618113.26][CONN][RXD] >>> Running case #1: 'Test: LowPowerTimer - stopped after creation.'...
[1542618113.36][CONN][INF] found KV pair in stream: {{__testcase_start;Test: LowPowerTimer - stopped after creation.}}, queued...
[1542618113.46][CONN][RXD] >>> 'Test: LowPowerTimer - stopped after creation.': 1 passed, 0 failed
[1542618113.46][CONN][RXD] 
[1542618113.46][CONN][INF] found KV pair in stream: {{__testcase_finish;Test: LowPowerTimer - stopped after creation.;1;0}}, queued...
[1542618113.56][CONN][RXD] >>> Running case #2: 'Test: LowPowerTimer - measure time accumulation.'...
[1542618113.66][CONN][RXD] :123::FAIL: Expected 0.010000 Was 0.000000
[1542618113.66][CONN][INF] found KV pair in stream: {{__testcase_start;Test: LowPowerTimer - measure time accumulation.}}, queued...
[1542618113.91][CONN][INF] found KV pair in stream: {{__testcase_finish;Test: LowPowerTimer - measure time accumulation.;0;1}}, queued...
[1542618113.91][CONN][RXD] >>> 'Test: LowPowerTimer - measure time accumulation.': 0 passed, 1 failed with reason 'Assertion Failed'
[1542618113.91][CONN][RXD] 
[1542618113.99][CONN][RXD] >>> Test cases: 1 passed, 1 failed with reason 'Assertion Failed'
[1542618113.99][CONN][RXD] >>> TESTS FAILED!
[1542618114.07][CONN][INF] found KV pair in stream: {{__testcase_summary;1;1}}, queued...
[1542618114.07][CONN][INF] found KV pair in stream: {{max_heap_usage;0}}, queued...
[1542618114.07][CONN][INF] found KV pair in stream: {{reserved_heap;185120}}, queued...
[1542618114.07][CONN][INF] found KV pair in stream: {{__thread_info;"0x0200025d0",708,4096}}, queued...
[1542618114.07][HTST][ERR] orphan event in main phase: {{max_heap_usage;0}}, timestamp=1542618114.073569
[1542618114.07][HTST][ERR] orphan event in main phase: {{reserved_heap;185120}}, timestamp=1542618114.073573
[1542618114.07][HTST][ERR] orphan event in main phase: {{__thread_info;"0x0200025d0",708,4096}}, timestamp=1542618114.073577
[1542618114.16][CONN][INF] found KV pair in stream: {{__thread_info;"0x0200002b4",64,512}}, queued...
[1542618114.16][CONN][INF] found KV pair in stream: {{__thread_info;"0x0200002f8",80,768}}, queued...
[1542618114.16][HTST][ERR] orphan event in main phase: {{__thread_info;"0x0200002b4",64,512}}, timestamp=1542618114.159263
[1542618114.16][HTST][ERR] orphan event in main phase: {{__thread_info;"0x0200002f8",80,768}}, timestamp=1542618114.159270
[1542618114.26][CONN][RXD] {{__cpu_info        up time;0}}
[1542618114.26][CONN][RXD] {{__cpu_info     sleep time;0}}
[1542618114.26][CONN][RXD] {{__cpu_info deepsleep time;0}}
[1542618114.36][CONN][RXD] {{__cpu_info  %  sleep/deep;0;0}}
[1542618114.36][CONN][INF] found KV pair in stream: {{end;failure}}, queued...
[1542618114.36][CONN][INF] found KV pair in stream: {{__exit;0}}, queued...
[1542618114.36][HTST][INF] __exit(0)
[1542618114.36][HTST][INF] __notify_complete(False)
[1542618114.36][HTST][INF] __exit_event_queue received
[1542618114.36][HTST][INF] test suite run finished after 1.80 sec...
[1542618114.36][CONN][INF] received special event '__host_test_finished' value='True', finishing
[1542618114.36][urllib3.connectionpool]Starting new HTTP connection (1): austin-ci-linux-001:8000
[1542618114.40][urllib3.connectionpool]http://austin-ci-linux-001:8000 "PUT /resource/0240000040214e4500341018c892003e9e11000097969900/disconnect HTTP/1.1" 200 2
[1542618114.40][urllib3.connectionpool]Starting new HTTP connection (1): austin-ci-linux-001:8000
[1542618117.56][urllib3.connectionpool]http://austin-ci-linux-001:8000 "PUT /resource/0240000040214e4500341018c892003e9e11000097969900/release HTTP/1.1" 200 66
[1542618117.56][HTST][INF] CONN exited with code: 0
[1542618117.56][HTST][INF] Some events in queue
[1542618117.56][HTST][INF] stopped consuming events
[1542618117.56][HTST][INF] host test result() call skipped, received: False
[1542618117.56][HTST][INF] calling blocking teardown()
[1542618117.56][HTST][INF] teardown() finished
[1542618117.56][HTST][INF] {{result;failure}}
```

### Issue request type

    [ ] Question
    [ ] Enhancement
    [ X ] Bug

0xc0170, 2018-11-20:
cc @ARMmbed/team-nxp

studavekar, 2018-12-03:
Mbed os hardware tests are passing in CI closing this issue.

closed 2018-12-03 by studavekar