ARMmbed/mbed-os #8806: tests-mbed_drivers-lp_timeout test fails on K64F - unstable

studavekar, 2018-11-19:
### Description

This failure is seen multiple times in CI system, created an issue to track it. May be worn out flash? 

**Target**
K64F

**Toolchain:**
GCC_ARM| ARM|IAR

**mbed-os sha:**

9aef9d3 Merge pull request #8671 from mray19027/uart_break

**Steps to reproduce**

Use command run :  mbed test -m K64F -t GCC_ARM  -n tests-mbed_drivers-lp_timeout  -v

Failure 

```
[1542619326.98][CONN][INF] sending up to 10 __sync packets (specified with --sync=10)
[1542619326.98][CONN][INF] sending preamble 'dc450103-7dcb-41dd-a6cc-2aee71494626'
[1542619326.98][GLRM][TXD] {{__sync;dc450103-7dcb-41dd-a6cc-2aee71494626}}
[1542619328.15][CONN][RXD] mbedmbedmbedmbedmbedmbedmbedmbed
[1542619328.15][CONN][INF] found SYNC in stream: {{__sync;dc450103-7dcb-41dd-a6cc-2aee71494626}} it is #0 sent, queued...
[1542619328.15][HTST][INF] sync KV found, uuid=dc450103-7dcb-41dd-a6cc-2aee71494626, timestamp=1542619328.151798
[1542619328.24][CONN][INF] found KV pair in stream: {{__version;1.3.0}}, queued...
[1542619328.24][CONN][INF] found KV pair in stream: {{__timeout;240}}, queued...
[1542619328.24][CONN][INF] found KV pair in stream: {{__host_test_name;timing_drift_auto}}, queued...
[1542619328.24][HTST][INF] DUT greentea-client version: 1.3.0
[1542619328.24][CONN][INF] found KV pair in stream: {{__testcase_count;22}}, queued...
[1542619328.24][HTST][INF] setting timeout to: 240 sec
[1542619328.24][HTST][INF] host test class: '<class 'timing_drift_auto.TimingDriftSync'>'
[1542619328.24][HTST][INF] host test setup() call...
[1542619328.24][HTST][INF] CALLBACKs updated
[1542619328.24][HTST][INF] host test detected: timing_drift_auto
[1542619328.34][CONN][RXD] >>> Running 22 test cases...
[1542619328.34][CONN][INF] found KV pair in stream: {{__testcase_name;Callback called once (attach)}}, queued...
[1542619328.44][CONN][INF] found KV pair in stream: {{__testcase_name;Callback called once (attach_us)}}, queued...
[1542619328.44][CONN][INF] found KV pair in stream: {{__testcase_name;Callback not called when cancelled (attach)}}, queued...
[1542619328.54][CONN][INF] found KV pair in stream: {{__testcase_name;Callback not called when cancelled (attach_us)}}, queued...
[1542619328.64][CONN][INF] found KV pair in stream: {{__testcase_name;Callback override (attach)}}, queued...
[1542619328.64][CONN][INF] found KV pair in stream: {{__testcase_name;Callback override (attach_us)}}, queued...
[1542619328.74][CONN][INF] found KV pair in stream: {{__testcase_name;Multiple timeouts running in parallel (attach)}}, queued...
[1542619328.84][CONN][INF] found KV pair in stream: {{__testcase_name;Multiple timeouts running in parallel (attach_us)}}, queued...
[1542619328.84][CONN][INF] found KV pair in stream: {{__testcase_name;Zero delay (attach)}}, queued...
[1542619328.94][CONN][INF] found KV pair in stream: {{__testcase_name;Zero delay (attach_us)}}, queued...
[1542619328.94][CONN][INF] found KV pair in stream: {{__testcase_name;10 ms delay accuracy (attach)}}, queued...
[1542619329.04][CONN][INF] found KV pair in stream: {{__testcase_name;10 ms delay accuracy (attach_us)}}, queued...
[1542619329.04][CONN][INF] found KV pair in stream: {{__testcase_name;1 s delay accuracy (attach)}}, queued...
[1542619329.14][CONN][INF] found KV pair in stream: {{__testcase_name;1 s delay accuracy (attach_us)}}, queued...
[1542619329.14][CONN][INF] found KV pair in stream: {{__testcase_name;5 s delay accuracy (attach)}}, queued...
[1542619329.24][CONN][INF] found KV pair in stream: {{__testcase_name;5 s delay accuracy (attach_us)}}, queued...
[1542619329.24][CONN][INF] found KV pair in stream: {{__testcase_name;1 s delay during sleep (attach)}}, queued...
[1542619329.34][CONN][INF] found KV pair in stream: {{__testcase_name;1 s delay during sleep (attach_us)}}, queued...
[1542619329.34][CONN][INF] found KV pair in stream: {{__testcase_name;1 s delay during deepsleep (attach)}}, queued...
[1542619329.63][CONN][RXD] 
[1542619329.64][CONN][INF] found KV pair in stream: {{__testcase_name;1 s delay during deepsleep (attach_us)}}, queued...
[1542619329.64][CONN][INF] found KV pair in stream: {{__testcase_name;Timing drift (attach)}}, queued...
[1542619329.64][CONN][INF] found KV pair in stream: {{__testcase_name;Timing drift (attach_us)}}, queued...
[1542619329.68][CONN][RXD] >>> Running case #1: 'Callback called once (attach)'...
[1542619329.68][CONN][INF] found KV pair in stream: {{__testcase_start;Callback called once (attach)}}, queued...
[1542619329.75][CONN][RXD] :85::FAIL: Expected 1 Was 0
[1542619329.75][CONN][INF] found KV pair in stream: {{__testcase_finish;Callback called once (attach);0;1}}, queued...
[1542619329.84][CONN][RXD] >>> 'Callback called once (attach)': 0 passed, 1 failed with reason 'Assertion Failed'
[1542619329.84][CONN][RXD] 
[1542619329.94][CONN][RXD] >>> Test cases: 0 passed, 1 failed with reason 'Assertion Failed'
[1542619329.94][CONN][RXD] >>> TESTS FAILED!
[1542619329.94][CONN][INF] found KV pair in stream: {{__testcase_summary;0;1}}, queued...
[1542619329.94][CONN][INF] found KV pair in stream: {{max_heap_usage;0}}, queued...
[1542619329.94][HTST][ERR] orphan event in main phase: {{max_heap_usage;0}}, timestamp=1542619329.937949
[1542619330.04][CONN][INF] found KV pair in stream: {{reserved_heap;184752}}, queued...
[1542619330.04][CONN][INF] found KV pair in stream: {{__thread_info;"0x020002740",656,4096}}, queued...
[1542619330.04][HTST][ERR] orphan event in main phase: {{reserved_heap;184752}}, timestamp=1542619330.039030
[1542619330.04][HTST][ERR] orphan event in main phase: {{__thread_info;"0x020002740",656,4096}}, timestamp=1542619330.039039
[1542619330.14][CONN][RXD] {{__cpu_info        up time;0}}
[1542619330.14][CONN][INF] found KV pair in stream: {{__thread_info;"0x0200002b4",80,512}}, queued...
[1542619330.14][CONN][INF] found KV pair in stream: {{__thread_info;"0x0200002f8",80,768}}, queued...
[1542619330.14][HTST][ERR] orphan event in main phase: {{__thread_info;"0x0200002b4",80,512}}, timestamp=1542619330.138465
[1542619330.14][HTST][ERR] orphan event in main phase: {{__thread_info;"0x0200002f8",80,768}}, timestamp=1542619330.138474
[1542619330.24][CONN][RXD] {{__cpu_info     sleep time;0}}
[1542619330.24][CONN][RXD] {{__cpu_info deepsleep time;0}}
[1542619330.24][CONN][RXD] {{__cpu_info  %  sleep/deep;0;0}}
[1542619330.34][CONN][INF] found KV pair in stream: {{end;failure}}, queued...
[1542619330.34][CONN][INF] found KV pair in stream: {{__exit;0}}, queued...
[1542619330.34][HTST][INF] __exit(0)
[1542619330.34][HTST][INF] __notify_complete(False)
[1542619330.34][HTST][INF] __exit_event_queue received
[1542619330.34][HTST][INF] test suite run finished after 2.10 sec...
[1542619330.34][CONN][INF] received special event '__host_test_finished' value='True', finishing
[1542619330.34][urllib3.connectionpool]Starting new HTTP connection (1): austin-ci-linux-001:8000
[1542619330.36][urllib3.connectionpool]http://austin-ci-linux-001:8000 "PUT /resource/0240000040214e4500341018c892003e9e11000097969900/disconnect HTTP/1.1" 200 2
[1542619330.36][urllib3.connectionpool]Starting new HTTP connection (1): austin-ci-linux-001:8000
[1542619333.53][urllib3.connectionpool]http://austin-ci-linux-001:8000 "PUT /resource/0240000040214e4500341018c892003e9e11000097969900/release HTTP/1.1" 200 66
[1542619333.53][HTST][INF] CONN exited with code: 0
[1542619333.53][HTST][INF] Some events in queue
[1542619333.53][HTST][INF] stopped consuming events
[1542619333.53][HTST][INF] host test result() call skipped, received: False
[1542619333.53][HTST][INF] calling blocking teardown()
[1542619333.53][HTST][INF] teardown() finished
[1542619333.53][HTST][INF] {{result;failure}}
```

### Issue request type

    [ ] Question
    [ ] Enhancement
    [ X ] Bug

mmahadevan108, 2018-11-20:
I don't see this failure. I ran the tests against commit 9aef9d36612ea4f75f93d254ee442ded0a1ffde0

Test results below:

mbedgt: running 3 tests for platform 'K64F' and toolchain 'ARM'
mbedgt: mbed-host-test-runner: started
mbedgt: checking for GCOV data...
mbedgt: test on hardware with target id: 0240000025354e450043400d47d00042bb91000097969900
mbedgt: test suite 'mbed-os-tests-mbed_drivers-lp_timeout' ........................................... OK in 56.32 sec
        test case: '1 s delay accuracy (attach)' ..................................................... OK in 1.06 sec
        test case: '1 s delay accuracy (attach_us)' .................................................. OK in 1.06 sec
        test case: '1 s delay during deepsleep (attach)' ............................................. OK in 1.08 sec
        test case: '1 s delay during deepsleep (attach_us)' .......................................... OK in 1.09 sec
        test case: '1 s delay during sleep (attach)' ................................................. OK in 1.06 sec
        test case: '1 s delay during sleep (attach_us)' .............................................. OK in 1.06 sec
        test case: '10 ms delay accuracy (attach)' ................................................... OK in 0.06 sec
        test case: '10 ms delay accuracy (attach_us)' ................................................ OK in 0.07 sec
        test case: '5 s delay accuracy (attach)' ..................................................... OK in 5.06 sec
        test case: '5 s delay accuracy (attach_us)' .................................................. OK in 5.05 sec
        test case: 'Callback called once (attach)' ................................................... OK in 0.07 sec
        test case: 'Callback called once (attach_us)' ................................................ OK in 0.09 sec
        test case: 'Callback not called when cancelled (attach)' ..................................... OK in 0.09 sec
        test case: 'Callback not called when cancelled (attach_us)' .................................. OK in 0.10 sec
        test case: 'Callback override (attach)' ...................................................... OK in 0.09 sec
        test case: 'Callback override (attach_us)' ................................................... OK in 0.09 sec
        test case: 'Multiple timeouts running in parallel (attach)' .................................. OK in 0.08 sec
        test case: 'Multiple timeouts running in parallel (attach_us)' ............................... OK in 0.08 sec
        test case: 'Timing drift (attach)' ........................................................... OK in 9.94 sec
        test case: 'Timing drift (attach_us)' ........................................................ OK in 9.39 sec
        test case: 'Zero delay (attach)' ............................................................. OK in 0.05 sec
        test case: 'Zero delay (attach_us)' .......................................................... OK in 0.06 sec
mbedgt: test case summary: 22 passes, 0 failures
mbedgt: mbed-host-test-runner: started
mbedgt: checking for GCOV data...
mbedgt: test on hardware with target id: 0240000025354e450043400d47d00042bb91000097969900
mbedgt: test suite 'mbed-os-tests-mbed_drivers-lp_ticker' ............................................ OK in 20.77 sec
        test case: 'Test attach for 0.001s and time measure' ......................................... OK in 0.07 sec
        test case: 'Test attach for 0.01s and time measure' .......................................... OK in 0.08 sec
        test case: 'Test attach for 0.1s and time measure' ........................................... OK in 0.16 sec
        test case: 'Test attach for 0.5s and time measure' ........................................... OK in 0.57 sec
        test case: 'Test attach_us for 100ms and time measure' ....................................... OK in 0.17 sec
        test case: 'Test attach_us for 10ms and time measure' ........................................ OK in 0.08 sec
        test case: 'Test attach_us for 1ms and time measure' ......................................... OK in 0.07 sec
        test case: 'Test attach_us for 500ms and time measure' ....................................... OK in 0.57 sec
        test case: 'Test detach' ..................................................................... OK in 0.74 sec
        test case: 'Test multi call and time measure' ................................................ OK in 1.06 sec
        test case: 'Test multi ticker' ............................................................... OK in 0.28 sec
mbedgt: test case summary: 11 passes, 0 failures
mbedgt: mbed-host-test-runner: started
mbedgt: checking for GCOV data...
mbedgt: test on hardware with target id: 0240000025354e450043400d47d00042bb91000097969900
mbedgt: test suite 'mbed-os-tests-mbed_drivers-lp_timer' ............................................. OK in 19.40 sec
        test case: 'Test: LowPowerTimer - float operator.' ........................................... OK in 0.06 sec
        test case: 'Test: LowPowerTimer - measure time accumulation.' ................................ OK in 1.24 sec
        test case: 'Test: LowPowerTimer - reset.' .................................................... OK in 0.08 sec
        test case: 'Test: LowPowerTimer - start started timer.' ...................................... OK in 0.10 sec
[... cut ...]
| K64F-ARM | K64F          | mbed-os-tests-mbed_drivers-lp_timer   | Test: LowPowerTimer - time measurement 10 ms.     |
 1      | 0      | OK     | 0.08               |
| K64F-ARM | K64F          | mbed-os-tests-mbed_drivers-lp_timer   | Test: LowPowerTimer - time measurement 100 ms.    |
 1      | 0      | OK     | 0.17               |
+----------+---------------+---------------------------------------+---------------------------------------------------+
--------+--------+--------+--------------------+
mbedgt: test case results: 42 OK
mbedgt: completed in 100.47 sec
PS C:\mymbed>

studavekar, 2018-12-03:
Mbed os hardware tests are passing in CI closing this issue.

closed 2018-12-03 by studavekar