ARMmbed/mbed-os #8808: tests-mbed_platform-stats_cpu test fails on K64F

studavekar, 2018-11-19:
### Description

This failure is seen multiple times in CI system, created an issue to track it. May be worn out flash? 

**Target**
K64F

**Toolchain:**
GCC_ARM| ARM|IAR

**mbed-os sha:**

9aef9d3 Merge pull request #8671 from mray19027/uart_break

**Steps to reproduce**

Use command run :  mbed test -m K64F -t GCC_ARM  -n tests-mbed_platform-stats_cpu  -v

Failure 

```
[1542617743.70][CONN][INF] found SYNC in stream: {{__sync;d5e8776e-801d-4489-86b9-dc5b6d7ee8dd}} it is #0 sent, queued...
[1542617743.70][HTST][INF] sync KV found, uuid=d5e8776e-801d-4489-86b9-dc5b6d7ee8dd, timestamp=1542617743.704033
[1542617743.81][CONN][INF] found KV pair in stream: {{__version;1.3.0}}, queued...
[1542617743.81][CONN][INF] found KV pair in stream: {{__timeout;20}}, queued...
[1542617743.81][CONN][INF] found KV pair in stream: {{__host_test_name;default_auto}}, queued...
[1542617743.81][HTST][INF] DUT greentea-client version: 1.3.0
[1542617743.81][HTST][INF] setting timeout to: 20 sec
[1542617743.81][CONN][INF] found KV pair in stream: {{__testcase_count;2}}, queued...
[1542617743.81][HTST][INF] host test class: '<class 'mbed_host_tests.host_tests.default_auto.DefaultAuto'>'
[1542617743.81][HTST][INF] host test setup() call...
[1542617743.81][HTST][INF] CALLBACKs updated
[1542617743.81][HTST][INF] host test detected: default_auto
[1542617743.91][CONN][RXD] >>> Running 2 test cases...
[1542617743.91][CONN][RXD] 
[1542617743.91][CONN][INF] found KV pair in stream: {{__testcase_name;Test CPU Info}}, queued...
[1542617743.91][CONN][INF] found KV pair in stream: {{__testcase_name;Test CPU load}}, queued...
[1542617744.00][CONN][RXD] >>> Running case #1: 'Test CPU Info'...
[1542617744.00][CONN][RXD] :70::FAIL: Expected Not-Equal
[1542617744.01][CONN][INF] found KV pair in stream: {{__testcase_start;Test CPU Info}}, queued...
[1542617744.11][CONN][INF] found KV pair in stream: {{__testcase_finish;Test CPU Info;0;1}}, queued...
[1542617744.21][CONN][RXD] >>> 'Test CPU Info': 0 passed, 1 failed with reason 'Assertion Failed'
[1542617744.21][CONN][RXD] 
[1542617744.21][CONN][RXD] >>> Test cases: 0 passed, 1 failed with reason 'Assertion Failed'
[1542617744.21][CONN][RXD] >>> TESTS FAILED!
[1542617744.30][CONN][INF] found KV pair in stream: {{__testcase_summary;0;1}}, queued...
[1542617744.31][CONN][INF] found KV pair in stream: {{max_heap_usage;0}}, queued...
[1542617744.31][CONN][INF] found KV pair in stream: {{reserved_heap;185296}}, queued...
[1542617744.31][HTST][ERR] orphan event in main phase: {{max_heap_usage;0}}, timestamp=1542617744.304906
[1542617744.31][HTST][ERR] orphan event in main phase: {{reserved_heap;185296}}, timestamp=1542617744.304910
[1542617744.41][CONN][INF] found KV pair in stream: {{__thread_info;"0x020002520",560,4096}}, queued...
[1542617744.41][CONN][INF] found KV pair in stream: {{__thread_info;"0x0200002b4",80,512}}, queued...
[1542617744.41][HTST][ERR] orphan event in main phase: {{__thread_info;"0x020002520",560,4096}}, timestamp=1542617744.405698
[1542617744.41][HTST][ERR] orphan event in main phase: {{__thread_info;"0x0200002b4",80,512}}, timestamp=1542617744.405710
[1542617744.67][CONN][RXD] {{__cpu_info        up time;0}}
[1542617744.67][CONN][RXD] {{__cpu_info     sleep time;0}}
[1542617744.67][CONN][INF] found KV pair in stream: {{__thread_info;"0x0200002f8",80,768}}, queued...
[1542617744.67][HTST][ERR] orphan event in main phase: {{__thread_info;"0x0200002f8",80,768}}, timestamp=1542617744.671326
[1542617744.67][CONN][RXD] {{__cpu_info deepsleep time;0}}
[1542617744.67][CONN][RXD] {{__cpu_info  %  sleep/deep;0;0}}
[1542617744.67][CONN][INF] found KV pair in stream: {{end;failure}}, queued...
[1542617744.67][CONN][INF] found KV pair in stream: {{__exit;0}}, queued...
[1542617744.68][HTST][INF] __exit(0)
[1542617744.68][HTST][INF] __notify_complete(False)
[1542617744.68][HTST][INF] __exit_event_queue received
[1542617744.68][HTST][INF] test suite run finished after 0.87 sec...
[1542617744.68][CONN][INF] received special event '__host_test_finished' value='True', finishing
[1542617744.68][urllib3.connectionpool]Starting new HTTP connection (1): austin-ci-linux-001:8000
[1542617744.73][urllib3.connectionpool]http://austin-ci-linux-001:8000 "PUT /resource/0240000040214e4500341018c892003e9e11000097969900/disconnect HTTP/1.1" 200 2
[1542617744.73][urllib3.connectionpool]Starting new HTTP connection (1): austin-ci-linux-001:8000
[1542617747.88][urllib3.connectionpool]http://austin-ci-linux-001:8000 "PUT /resource/0240000040214e4500341018c892003e9e11000097969900/release HTTP/1.1" 200 66
[1542617747.89][HTST][INF] CONN exited with code: 0
[1542617747.89][HTST][INF] Some events in queue
[1542617747.89][HTST][INF] stopped consuming events
[1542617747.89][HTST][INF] host test result() call skipped, received: False
[1542617747.89][HTST][INF] calling blocking teardown()
[1542617747.89][HTST][INF] teardown() finished
[1542617747.89][HTST][INF] {{result;failure}}
```

### Issue request type

    [ ] Question
    [ ] Enhancement
    [ X ] Bug

0xc0170, 2018-11-20:
cc @ARMmbed/team-nxp

studavekar, 2018-11-20:
We have the device in failing state and it been debugged currently.

studavekar, 2018-12-03:
Mbed os hardware tests are passing in CI closing this issue.

closed 2018-12-03 by studavekar