ARMmbed/mbed-os #8816: mbed assert in Thread::_task when setting flags before Thread::start() is finished

BramdeBoer1, 2018-11-20:
### Description
Target: STM32L072
IDE: IAR 8.30.1
Mbed OS: 5.10.3

Starting a new thread with a higher priotity than the thread which started the new thread and setting flags (before the osThreadNew returns) will cause an mbed assert on a missing Thread::_tid.

The root cause is a thread switch before Thread::start() is finished and not waiting for start to finish execution.

In Thread.ccp::113 the _tid member is assigned
`_tid = osThreadNew(Thread::_thunk, this, &_attr);`

Before the assignement the following code causes a thread switch
In rtx_thread.c:824
```
if (thread != NULL) {
    osRtxThreadDispatch(thread);
  }
```

This causes Thread::_thunk to be invoked in Thread.cpp:428.

Now doing anything which needs the _tid member before it is actually set will lead to an immediate mbed assert.

Code to reproduce:
```
#include "Thread.h"

#define STACK_SIZE 256
class CApp : public rtos::Thread
{
public:
    CApp();
    void mStart()       { start( this, &CApp::m_Task ); }

private:
    void                m_Task();
    uint64_t            m_ThreadStack[ STACK_SIZE / sizeof(uint64_t) ];
};

CApp::CApp()
:   Thread( osPriorityAboveNormal, APP_STACK_SIZE, reinterpret_cast<uint8_t *>(m_ThreadStack), "App Thread" ),
    m_ThreadStack()
{
}

void CApp::m_Task()
{
    while( true )
    {
// This will cause a mbed assert
        flags_set( 1 );
    }
}

CApp app;
int main()
{
    app.mStart();
}
```

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

BramdeBoer1, 2018-11-20:
I managed to solve it by adding a Semphore to the Thread thus preventing it from running any task code before completing the start.
I'm however concerened that the increase in footprint might be overkill for a situation which should only occur once.
Maybe there are better ways to solve this issue.

In Thread.h added:
Semaphore                  _start_sem;

In Thread.cpp changed the following:
```
osStatus Thread::start(Callback<void()> task)
{
    _mutex.lock();

    if ((_tid != 0) || _finished) {
        _mutex.unlock();
        return osErrorParameter;
    }

    if (_attr.stack_mem == NULL) {
        _attr.stack_mem = new uint32_t[_attr.stack_size / sizeof(uint32_t)];
        MBED_ASSERT(_attr.stack_mem != NULL);
    }

    //Fill the stack with a magic word for maximum usage checking
    for (uint32_t i = 0; i < (_attr.stack_size / sizeof(uint32_t)); i++) {
        ((uint32_t *)_attr.stack_mem)[i] = osRtxStackMagicWord;
    }

    memset(&_obj_mem, 0, sizeof(_obj_mem));
    _attr.cb_size = sizeof(_obj_mem);
    _attr.cb_mem = &_obj_mem;
    _task = task;
    _tid = osThreadNew(Thread::_thunk, this, &_attr);
    if (_tid == NULL) {
        if (_dynamic_stack) {
            delete[](uint32_t *)(_attr.stack_mem);
            _attr.stack_mem = (uint32_t *)NULL;
        }
        _mutex.unlock();
        _join_sem.release();
        // Release
        _start_sem.release();
        return osErrorResource;
    }

    _mutex.unlock();
    // Release
    _start_sem.release();
    return osOK;
}

void Thread::_thunk(void *thread_ptr)
{
    Thread *t = (Thread *)thread_ptr;
    // Wait
    t->_start_sem.wait();
    t->_task();
    t->_mutex.lock();
    t->_tid = (osThreadId)NULL;
    t->_finished = true;
    t->_join_sem.release();
    // rtos will release the mutex automatically
}

```

evedon, 2019-05-24:
There has been no activity for a while. 
@BramdeBoer1  Is this still an issue with Mbed OS 5.12.4?

BramdeBoer1, 2019-06-06:
Sorry for the late reply.

I have just verified that with mbed os 5.12.4 this issue still persist.

Tested the issue with the code to reproduce as described above.

closed 2021-10-20 by ciarmcom